prusa3d/PrusaSlicer #6055: Gaps between walls

Nicolette14, 2021-02-15:
Hi guys,

I am very new to 3D printing. I have made multiple designs which slice just fine. But some of by designs have gaps on the thinnest layer after slicing in PrusaSlicer (Slic3r). 

How do I fix this? I have to check the box print thin walls otherwise there are even bigger gaps. I think the problem is that the line is to thinn at that point. But I don't know how I can fix this so it is a clean line with the right width.
[image]

On some other walls I get these weird lines/crossings.
[image]

The design is as followed:
[image]

Thank you in advance for the help!

rtyr, 2021-02-15:
Please provide your 3mf project file.

Nicolette14, 2021-02-15:
Sorry, see attachment
[Moroccan chandelier set.zip](https://github.com/prusa3d/PrusaSlicer/files/5982060/Moroccan.chandelier.set.zip)

rtyr, 2021-02-17:
The model has very thin walls (+ variable level of thickness), too thin to be printed with your extrusion width. It is somehow printable with "detect thin walls" (with the defects on your screenshots), but the proper solution would be to modify the model (or at least make it bigger). PrusaSlicer tries to squeeze one or two lines into the ring. It is on the threshold of placing one or two lines. While we plan to improve the behavior in such cases, this is imho mainly caused  by non optimal model.

closed 2021-03-04 by rtyr