prusa3d/PrusaSlicer #6058: support missing under horizontal overhang

zxcv-org, 2021-02-15:
### Version
2.3.0+win64

### Operating system type + version
Windows 10 Pro
64 bit

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S+

### Behavior
Supports are not generated under a huge horizontal overhang.  I've increased the raft layers to make it super obvious, but it still occurs with less raft, and I think also occurs with no raft, but it's more subtle with no raft.

Obviously this wouldn't print successfully (I didn't try, but ... well it wouldn't).

Expected results:
Support should be generated under the horizontal overhang.

Actual results:
Support is missing under horizontal overhang.

I believe this is a bug, not a new feature request.

#### Project File (.3MF) where problem occurs
[image]
[borken supports -- book_box_casket_.zip](https://github.com/prusa3d/PrusaSlicer/files/5984358/borken.supports.--.book_box_casket_.zip)

bubnikv, 2021-02-19:
It looks like it has something to do with the raft and possibly suppression of supports below bridges.

bubnikv, 2021-02-24:
It seems to be fixed on current master.

[image]

zxcv-org, 2021-03-03:
I just cloned master and built locally, and I still see the problem, in both Release and Debug builds.

My master is at this commit:

commit 91ffd8d50167bc987ce44be93df651b82613a9ba (HEAD -> master, origin/master, origin/HEAD)
Author: Enrico Turri <enricoturri@seznam.cz>
Date:   Tue Mar 2 17:00:11 2021 +0100

    Custom G-code validation - Do not veto setting tab switch when invalid g-code is found

zxcv-org, 2021-03-03:
[image]

NextDesign1, 2021-04-03:
Do you have the option "Don't support bridges" enabled? I ran into the same issue and disabling it seemed to fix it.

zxcv-org, 2021-04-06:
> Do you have the option "Don't support bridges" enabled? I ran into the same issue and disabling it seemed to fix it.

"Don't support bridges" unchecked - as shown in previously-attached image.
"Don't support bridges" checked - as shown in previously-attached image.

bubnikv, 2021-04-07:
I have a fix in the working.

bubnikv, 2021-04-08:
Fixed with 8fd731f

zxcv-org, 2021-08-01:
Happening (still? / again?) as of a54f5fb41a5d9c4301d217490c086ca6ae208246, which is later than 8fd731f.

rtyr, 2021-08-02:
Reproduced on current master.

bubnikv, 2021-10-04:
Hopefully fixed for good with 6192c573682d1a817ce577625af9e840b2f53fc0

    There was a bug for more than 4 years, which likely triggers now more
    often with the introduction of raft_contact_distance parameter,
    which is usually significantly smaller than support_material_contact_distance.
    There were no support towers built at the raft for contact layers
    below the print_z of the first object layer.

closed 2021-10-04 by bubnikv