prusa3d/PrusaSlicer #6059: Place on Face behavior change

andrewluebke, 2021-02-16:
### Version
2.3.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa SL1

### Behavior
In 2.2 when when using the place on face the slicer appeared to rotate the model on the X-axis by 180 degrees and place on the surface.  With 2.3 it appears to rotate around the X-axis by 180 degrees and around the Y-axis by a random amount.  So if the model was inside the borders before the place on face once the operation is completed it will usually not be in the borders.  Not sure if this is by design but the old behavior was less work for me at least.

#### Project File (.3MF) where problem occurs
[image]
[image]

rtyr, 2021-02-17:
Can you share your project file?

bubnikv, 2021-02-17:
There was a change in the code, which may lead to such effects.

andrewluebke, 2021-02-18:
Sure here is an example before I lay it flat.  Once it is placed on a the top face it is at a strange angle.

[Example.zip](https://github.com/prusa3d/PrusaSlicer/files/5999427/Example.zip)

lukasmatena, 2021-02-18:
@andrewluebke Did you notice this problem for any other face but the top (i.e. horizontal) one?

andrewluebke, 2021-02-20:
No that is the face that I usually flip, I have two sets of teeth that are parallel.

bubnikv, 2021-09-25:
Nobody else chimed in. I suppose the current behavior is fine to most.

closed 2021-09-25 by bubnikv

sslupsky, 2021-10-05:
@bubnikv I have encountered this issue as well.  I prefer that the algorithm did not make the arbitrary rotation or if it does, that at a minimum the rotation property for the object is updated so it can be reset to 0.  Moreover, the object properties should be updated when using Place on Face regardless of this issue.

Place on Face does not work reliably either.  That is, I noticed that when using Place on Face the algorithm occasionally causes a small angle between the surface plane of the object and the surface plane of the bed thus the object is not flat on the bed.  In the screenshot below, you can see that the two objects in the lower left are not flat.

I have tested with 2.3.0 and 2.4.0-alpha3.

[image]

sslupsky, 2021-10-05:
@bubnikv The STL's for these two objects can be downloaded from Thingiverse:
https://www.thingiverse.com/thing:4938232
https://www.thingiverse.com/thing:4970586

These objects also demonstrate the arbitrary rotation angle issue.  To replicate the rotation use "Place on Face" on one of the two surfaces shown in the screenshot and then apply Place on Face to the top surface.

[image]

sslupsky, 2021-10-05:
The none planarity issue I mentioned might be related to Issue #5011.

Please reopen.

bubnikv, 2021-10-06:
The issue happens if the newly selected plane is parallel or nearly parallel to the previous plane. The rotation axis is chosen to minimize the rotation, but for these "degenerate" cases the chosen axis is just a tiny bit "better" than an axis aligned axis (aligned with world X or world Y). We may try to improve.

FrHePo, 2023-12-16:
I would like to be able to align the marked face exactly along the x-axis.

sslupsky, 2026-09-02:
Has this been addressed in version 3?