prusa3d/PrusaSlicer #6062: Bug? Slicer Does Not Pause At Height Creality CR 6

stratus-ss, 2021-02-16:
### Version
2.30

### Operating system type + version
```
System:
  Host: stratus-desktop Kernel: 5.4.97-1-lts x86_64 bits: 64 
  Desktop: GNOME 3.38.3 Distro: Arch Linux 
Machine:
  Type: Desktop System: ASUS product: All Series v: N/A 
  Mobo: ASUSTeK model: Z97-WS v: Rev 1.xx serial: <superuser required> 
  UEFI [Legacy]: American Megatrends v: 2403 date: 06/18/2015 
CPU:
  Info: Quad Core Intel Core i7-4790K [MT MCP] speed: 2666 MHz 
  min/max: 800/4400 MHz 
Graphics:
  Device-1: NVIDIA GM204 [GeForce GTX 970] driver: nvidia v: 460.39 
  Display: x11 server: X.Org 1.20.10 driver: loaded: nvidia 
  unloaded: modesetting,vesa resolution: 1: 1920x1080~60Hz 
  2: 2560x1440~100Hz 3: 1080x1920~60Hz 4: 1920x1080~60Hz 
  OpenGL: renderer: GeForce GTX 970/PCIe/SSE2 v: 4.6.0 NVIDIA 460.39 
```

### 3D printer brand / version + firmware version (if known)

Creality CR 6 SE  running official firmware: v 1.0.3.6

### Behavior

_Describe the problem_
When selecting the "Add Pause Print (M601) the printer ignores this code.
Using Cura software using the `BQ (M25)` pause settings works. I am not sure if this exists in the Prusa software or not

 _Steps needed to reproduce the problem_

Any time the pause is inserted into the gcode, the printer ignores it

_Expected Results_

Printer should pause at specified height/layer

_Actual Results_

Printer does not pause, the specified text is not displayed on the screen

_Is this a new feature request?_

Feature exists but does not function properly on Creality printer

#### Project File (.3MF) where problem occurs
[xyzCalibration_cube.zip](https://github.com/prusa3d/PrusaSlicer/files/5990260/xyzCalibration_cube.zip)

rtyr, 2021-02-16:
You can change the pause g-code to whatever command your printer supports (printer settings). Closing.

[image]

closed 2021-02-16 by rtyr

pmjdebruijn, 2021-02-17:
@stratus-ss: can you verify that `M25` as `pause_print_gcode` works for your CR-6 SE?

stratus-ss, 2021-02-17:
@pmjdebruijn I can confirm this works:

[image]

Can anyone tell me where to find the gcode to do a retraction is? I'd like to retract to happen when pause-at-height

pmjdebruijn, 2021-02-17:
@stratus-ss thanks! (hopefully this should become available by default in the future #6062)

The problem with retracting at pause, is that I don't think there is a way to unretract when resuming, but @rtyr might know better.

stratus-ss, 2021-02-17:
> @stratus-ss thanks! (hopefully this should become available by default in the future #6069)
> 
> The problem with retracting at pause, is that I don't think there is a way to unretract when resuming, but @rtyr might know better.

could we do a Z axis raise? The problem is some times when resuming, I suffer stringing on the print as the head moves to next position

rtyr, 2021-02-17:
Pause behavior is configured in the firmware.

pmjdebruijn, 2021-02-17:
M25 is a plain pause if I'm not mistaken.

In upstream Marlin M125 P1 is the "smart" pause, but most (factory) firmwares don't seem to have M125 enabled.