prusa3d/PrusaSlicer #6065: Why polygon point coordinates are stored as integer rather than floating point

Megidd, 2021-02-17:
The `Point` class always stores the coordinates as integer rather than floating point:

https://github.com/prusa3d/PrusaSlicer/blob/a116914fceb56bc6b8c8962db3589d021e31af52/src/libslic3r/Point.hpp#L91

Why is that? Are the Prusa slicer output polygon coordinates always in integer rather than floating point? 

To me, it seems counter-intuitive to *not* use floating point for storing coordinates. Not sure 🙂

foreachthing, 2021-02-17:
Floating points are a nightmare, if you need accurate numbers. Read more here: 
https://stackoverflow.com/questions/588004/is-floating-point-math-broken

In C#/VB.net, I often use decimals, if memory and cpu time is no issue.

Megidd, 2021-02-17:
@foreachthing Just to make sure I understand correctly, the floating-point to integer conversion is an approximation which we accept. Right?

lukasmatena, 2021-02-17:
@Megidd The problem with floating-point numbers is that you might get inconsistencies during the computations, such as when checking if a point lies on a line, if two vertices coincide and so on. That's why predictable fixed point arithmetic is used.

Quoting from https://github.com/junmer/clipper-lib/blob/master/Documentation.md:
> The IntPoint structure is used to represent all vertices in the Clipper Library. An "integer" storage type has been deliberately chosen to preserve numerical robustness. (Early versions of the library used floating point coordinates, but it became apparent that floating point imprecision would always cause occasional errors.)

The broken link in the quote probably linked to this:
https://cw.fel.cvut.cz/b181/_media/courses/cg/lectures/nonrobust_cgta_06.pdf

Have fun. Closing.

closed 2021-02-17 by lukasmatena

Megidd, 2021-02-17:
Just as a side note, I just figured out that the slicer apparently does one noticeable operation before converting `double` coordinates to `int`:

* Divide doubles by a `SCALING_FACTOR` of `0.000001`

`SCALING_FACTOR` is defined here and is used throughout the code:

https://github.com/prusa3d/PrusaSlicer/blob/a116914fceb56bc6b8c8962db3589d021e31af52/src/libslic3r/libslic3r.h#L49

foreachthing, 2021-02-17:
> * Divide doubles by a `SCALING_FACTOR` of `0.000001`

Well, you have to! 
Covert 0.01 to INT will result in zero. So, 0.01/0.000001=10000. Then you can muliply it by 1/0.000001 to get 0.01 again.