prusa3d/PrusaSlicer #6066: Ttemperature error in Ender-3 profile

petrik198, 2021-02-17:
### Version
_Version of PrusaSlicer 2.3.0

### Operating system Windows 10 Home 64bit
### 3D printer Ender 3
### Behavior
In the printer settings (in the section g-code itself) the hotend temperature on the M104 **### S120** is incorrectly entered! This is a Creality Ender-3 profile

[image]

rtyr, 2021-02-17:
It is intended. See https://github.com/prusa3d/PrusaSlicer/pull/5863.

rtyr, 2021-02-17:
FYI  @pmjdebruijn 

It seems I was right when I said it may be confusing.

pmjdebruijn, 2021-02-17:
@petrik198, the nozzle is temporarily heated to 120C, so it can warm up during homing/abl movements without any filament oozing out of the nozzle. After that the nozzle is moved to the front of the bed, where it's easy for the user to wick away any oozing filament as it continues heating to whatever temperature is requested. This approach basically reduces the printer start time.

@rtyr, given how ridiculously popular the Ender-3 is, and how long it took for the first report to show up, I'd argue the vast majority of users are coping just fine. More or less as I suspected.

While the start_gcode already has comments explaining this (as illustrated by the posted screenshot), I'll try to improve the wording a little bit.

As far as I'm concorned this issue can be closed as INVALID.

rtyr, 2021-02-17:
Ok, thanks. Closing.

closed 2021-02-17 by rtyr

pmjdebruijn, 2021-02-17:
For completeness sake: #6069