prusa3d/PrusaSlicer #6067: To much support

stonefred, 2021-02-17:
### Version

[image]
[image]

PrusaSlicer
Version:   2.3.0+win64
Build:     PrusaSlicer-2.3.0+win64-202101111315

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 34,304MB
OpenGL installation
GL version:   4.6.0 NVIDIA 456.71
Vendor:       NVIDIA Corporation
Renderer:     GeForce RTX 2080/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
[... 337 lines cut ...]
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### Behavior
If I change to 45° the result of the support is to big. 
Hint: 0° is OK.

_Is this a new feature request?_
I don't know.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
Sorry, but I connot uppload my file "Maske HP reverb g2   eSUN ABS Alleteile-A.3mf" (I get the message "We don't suppor that file Type" ?!).

Thanks Fritz

rtyr, 2021-02-17:
Please provide your 3MF project file. You have to zip it first.

stonefred, 2021-02-17:
[Maske HP reverb g2.zip](https://github.com/prusa3d/PrusaSlicer/files/5996895/Maske.HP.reverb.g2.zip)
Done.

stonefred, 2021-02-17:
It’s done.  Gesendet von Mail für Windows 10 Von: rtyrGesendet: Mittwoch, 17. Februar 2021 16:24An: prusa3d/PrusaSlicerCc: stonefred; AuthorBetreff: Re: [prusa3d/PrusaSlicer] To much support (#6067) Please provide your 3MF project file. You have to zip it first.

bubnikv, 2021-02-19:
Fix is on its way

[image]

bubnikv, 2021-02-19:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5209
Closing this one, it will be referenced from the other one.

closed 2021-02-19 by bubnikv