prusa3d/PrusaSlicer #6071: Dark mode on Ubuntu hides warning triangle

Kritten, 2021-02-17:
### Version
2.3.0
https://forum.prusaprinters.org/forum/prusaslicer/slicer-misses-surfaces/

### Operating system type + version
Ubuntu 20.04.2 LTS

### 3D printer brand / version + firmware version (if known)
not relevant

### Behavior
Due to the dark mode of PrusaSlice on Ubuntu, the small warning triangle next to an object in the right panel is not visible.
A description and Screenshot with the problem is available in this thread: https://forum.prusaprinters.org/forum/prusaslicer/slicer-misses-surfaces/#post-374365

[image]

Jan-Soustruznik, 2022-08-01:
Hello, Kritten,

Jan-Soustruznik, 2022-08-01:
Hello Kitten,

Could you check the issue with new released version of PrusaSlicer, please? 

Link: [PrusaSlicer 2.5.0-alpha3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.5.0-alpha3)

Thank you, have a nice day

Kritten, 2022-08-01:
Hi,

sadly it's still the same, hardly noticable:
[image]

closed 2026-09-29 by prusaslicer-bot[bot]