prusa3d/PrusaSlicer #6077: Slicing

telboy595, 2021-02-18:
### Version
_Version of PrusaSlicer used goes here_
2.30
_Use `About->About PrusaSlicer` for release versions_
Version 2.3.0+win64
_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Win 10
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa Mk3S with MMU 2

### Behavior
* _Describe the problem_
I have an aircraft to print using Prusaslicer and my printer as above, slicer imports the stl no problem and you then slice it to print and it misses a lot of the internal out. I have tried various set ups with no luck it does not like to print the internals 
I have noticed others on the forum have had the same issue
[Terrys Gcodes.zip](https://github.com/prusa3d/PrusaSlicer/files/6004228/Terrys.Gcodes.zip)

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-02-18:
Set slice gap closing radious to 0.

[image]

rtyr, 2021-02-21:
Answered, no response. Closing.

closed 2021-02-21 by rtyr

telboy595, 2021-02-22:
Sorry I checked over the weekend and problem still exists

Best regards

Terry Warner

KB Agency
Tel          (44)1205 481121
Mobile   (44) 7899 668079

DISCLAIMER:
The information in this E-Mail message is confidential and may be legally privileged. It is intended solely for the addressee and access to this message by anyone else is unauthorised. If you have received the message in error, please notify us immediately and delete it. If you are not the intended recipient, any disclosure, copying, distribution or any other use is prohibited and is be unlawful.

rtyr, 2021-02-22:
Seems OK to me, see the screenshot.

[image]

See also https://github.com/prusa3d/PrusaSlicer/issues/3062.

telboy595, 2021-02-23:
HI here is another zip of a 3dlab plane of  a section of the fus showing what happens when sliced 
[test files.zip](https://github.com/prusa3d/PrusaSlicer/files/6027963/test.files.zip) 

hope you find a solution

rtyr, 2021-02-23:
3Dlabprint models are problematic in general, there is an issue for that #3062.