prusa3d/PrusaSlicer #6080: Prusa Slic3r 2.3.0 interface is extremely SLOW in MacOS Big Sur

briandilley, 2021-02-18:
### Version
Version 2.3.0+

### Operating system type + version
MacOS 11.2 (Big Sur)
Mac Pro (Late 2013)
3GHz 8-core intel xeon E5
128 GB 1066 MHz DDR3
AMD FirePro D700 6 GB

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 Pro (although it's irellevant for this issue)

### Behavior
The interface is EXTREMELY slow. Clicking on interface elements results in a beachball (loading cursor) for a few seconds before anything happens. literally every interaction with the UI is this way.  I've tried disabling the retina setting for the 3d scene and it made that scene more responsive (ie: pan/tilt/zoom) but the rest of the UI/widgets are slow AF.

xyleth, 2021-02-18:
As another data point, I have a 2019 iMac running Big Sur and I don't have any issues with Prusa Slic3r running slowly, quite the contrary in fact.

I'm on:
MacOS 11.2.1
iMac Retina 5k, 27-Inch, 2019
3.7Ghz 6-Core Intel Core i5
40Gb 2667 Mhz DDR4 RAM
Radeon Pro 580X 8Gb Graphics

Running Version 2.3.0+ of Prusa Slicer.

bubnikv, 2021-02-19:
It has been fixed on the master, the upcoming release shall fix it.

hoeni, 2021-02-23:
Looking forward to the fix. Switching Tabs alone takes 1-2 seconds on a 8/16core iMac with 64 GB RAM :-)

jkotowicz1, 2021-03-06:
Same here, with bigger models basically 2.3.0 is basically unusable- take for ever to slice on MacBook Pro 13" 2015 running Catalina, no problems ever before. Sometimes it's stuck and doesn't slice at all. Hoping for a quick fix

briandilley, 2021-03-15:
The size of the models don't matter for me - it can literally have no model loaded whatsoever and it's still slow.

closed 2026-09-29 by prusaslicer-bot[bot]