prusa3d/PrusaSlicer #6081: Invalid Gcode generation

fri2, 2021-02-19:
### Version
Version 2.3.0+win64

### Operating system type + version
Version: 2.3.0+win64
Build: PrusaSlicer-2.3.0+win64-202101111315
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 34,054MB
WorkingSet: 397MB
PrivateBytes: 639MB
Pagefile(peak): 639MB(923MB)
RAM size reserved for the Undo / Redo stack: 1,074MB
RAM size occupied by the Undo / Redo stack: 14MB
OpenGL installation
GL version: 4.6.0 NVIDIA 452.56
Vendor: NVIDIA Corporation
Renderer: Quadro P1000/PCIe/SSE2

### 3D printer brand / version + firmware version (if known)
Prusa MK3s

### Behavior
* The Prusa Slicer is generating a badly looking Gcode, some layers are missing or are badly generated. 
* When using the Cura slicer Version 4.8.0, no problem occurs and the Gcode looks fine.

[image]

#### Project File (.3MF) where problem occurs
(impossible to join files with an .3MF ???? , got a "We don't support that file type" message from GitHUB)
==> So you will have to remove the .TXT at the end of the following file to obtain the .3MF file
[Bod 3.3MF.TXT](https://github.com/prusa3d/PrusaSlicer/files/6010192/Bod.3.3MF.TXT)

regards

fri2, 2021-02-19:
The STL file used :
(again, impossible to join files with a .stl extension ???? , got a "We don't support that file type" message from GitHUB)
so, remove the .TXT extension a the end of the following file 
[Bod 3.stl.TXT](https://github.com/prusa3d/PrusaSlicer/files/6010213/Bod.3.stl.TXT)

fri2, 2021-02-19:
Cura 4.8 gcode generation picture for the stl file :
[image]

neophyl, 2021-02-19:
That stl has the following errors -
Non manifold edges - 14 (these will be the biggest problem for PS)
Intersecting Faces - 22
Zero faces - 1007
Thin Faces - 37
Sharp Edges - 22

Report generated with Blenders 3dprint tool.

Here is a fixed version of your stl using the same tools Make manifold button.  Which I quickly sliced in PS2.3 and it looks ok to me but its not my model so I would check it fully.  It doesn't have the huge missing bits that the non manifold errros cause though.

[Bod.3_neo.zip](https://github.com/prusa3d/PrusaSlicer/files/6010296/Bod.3_neo.zip)

fri2, 2021-02-19:
Hello, 
yes I saw that the STL generated by Fusion360 was having errors in it.

neophyl, 2021-02-19:
If you are on windows 10 you can right click the model and use the fix through netfabb option which is built into PS already.

closed 2026-09-29 by prusaslicer-bot[bot]