prusa3d/PrusaSlicer #6083: Only first model extrudes

destinybonavita, 2021-02-19:
### Version
_2.3.0-rc3_

### Operating system type + version
_MacOS 11.2.1 (20D74)_

### 3D printer brand / version + firmware version (if known)
_AnyCubic Mega S_
Firmware: https://www.thingiverse.com/thing:3249319

### Behavior
* _I'm slicing [this model](https://thangs.com/ChaosCoreTech/Bundled-Up-Bernie-10524) with "Support on build plate only" and only the first layer is extruding when printing. I tested to see if I have this issue with other problems and I don't_
* _Steps needed to reproduce the problem_
  * _Add model to slicer_
  * _Use recommended profile for AnyCubic Mega S printer_
  * _Use recommended profile for generic PLA_
* _Expected Results_
  * _Each layer would extrude normally_
* _Actual Results_
  * _Only the initial layer is extruding while printing, the print head continues to move through the coordinates but the extruder stops extruding after the initial layer_

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
This [zip file](https://github.com/prusa3d/PrusaSlicer/files/6011262/Archive.zip) contains the 3mf file and the gcode that was produced when slicing.

rtyr, 2021-02-19:
Not PrusaSlicer related issue, the g-code is OK.

destinybonavita, 2021-02-19:
> Not PrusaSlicer related issue, the g-code is OK.

It looked okay to me too. It was really weird that I sliced this with Prusa, had the extrusion issue, sliced with Cura, no issue. Tried the same gcode again and same issue. Any idea what could cause this?

rtyr, 2021-02-19:
Probably firmware issue. I know people have often issues like that with incorrectly set TMC driver for the extruder. I can see that you are not using stock firmware, so I guess your hardware is also not stock.

rtyr, 2021-02-19:
Anyway, I printed few layers of your gcode...

[image]

destinybonavita, 2021-02-19:
> Anyway, I printed few layers of your gcode...
> 
> [image]

Really weird. That's the only model I'm having issues with in Prusa. I'm currently printing [this](https://www.thingiverse.com/thing:2719896) without any issues and I sliced it with Prusa

destinybonavita, 2021-02-19:
I'll have to keep tweaking to see what happened to this gcode after my current print completes. Thank you for looking into it!

closed 2021-02-19 by destinybonavita