prusa3d/PrusaSlicer #6084: Hints in the program do not work, how to enable?

AKlionRU, 2021-02-19:
### Version
2.3.0
### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
macOS Big Sur

Hints in the program do not work, how to enable? They work in the Windows version, there is no Mac version, why?
[image]

YuSanka, 2021-02-19:
I think you should ask at https://github.com/supermerill/SuperSlicer

closed 2021-02-19 by YuSanka

AKlionRU, 2021-02-19:
I showed the screenshot as an example, but I cannot show what is not there, the prbshasdaiser does not have this ...