prusa3d/PrusaSlicer #6089: temperature Macro in combination with next_extruder/previous_extruder does not work

Saij, 2021-02-19:
### Version
Version: 2.3.0+win64
Build: PrusaSlicer-2.3.0+win64-202101111315

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 17,110MB

### 3D printer brand / version + firmware version (if known)
Prusa MK3S (3.9.3) + MMU2S (1.0.6)

### Behavior
When using following toolchange G-Code, PrusaSlicer shows an error message about a wrong custom G-Code.

```
; TOOLCHANGE SPECIFIC CODE
; NEW EXTRUDER [next_extruder]
; NEW TEMP [temperature]
; TEMP 0 [temperature_0]
; TEMP 1 [temperature_1]
; TEMP 2 [temperature_2]
; TEMP 3 [temperature_3]
; TEMP 4 [temperature_4]
; TEST [temperature_[previous_extruder]]
; TEST [temperature_[next_extruder]]
```

Error Message:
```
G-code export to C:\Users\CHRIST~1\AppData\Local\Temp\.23960.gcode failed due to invalid custom G-code sections:
toolchange_gcode
Parsing error at line 9: Negative vector index
; TEST [temperature_[previous_extruder]]
        ^
Please inspect the file C:\Users\CHRIST~1\AppData\Local\Temp\.23960.gcode.tmp for error messages enclosed between
        !!!!! Failed to process the custom G-code template ...
and
        !!!!! End of an error report for the custom G-code template ...
for all macro processing errors.
```

I checked the page https://help.prusa3d.com/en/article/list-of-placeholders_205643/ if I maybe had a typo, but even a direct copy of the page won't work

cheeyy, 2021-02-24:
I am having the same issue!

bubnikv, 2021-02-24:
try

```

; TEST {temperature[previous_extruder]}
; TEST {temperature[next_extruder]}

```

Saij, 2021-02-25:
It works! Thank you!

closed 2021-02-25 by Saij