prusa3d/PrusaSlicer #6090: Feature Request:  Sloppy support

jrmoserbaltimore, 2021-02-19:
### Version
2.3.0

### 3D printer
FDM

### Behavior
Support flow cannot be adjusted (maybe there's a way to do this with a support enforcer?), and I want to experiment with sloppy supports.

My printer basically failed adhesion on the support material and dumped a pile of mangled filament in the corner, without damaging the print.  When it started extruding supports under a not-very-steep overhang, it spit out some kind of garbage (due to printing in mid air, of course) that was barely attached to the print.  This is a from-build-plate-only print.

This gave me a horrible overhang…for the support.  The actual print is fine.

So now I have a support kind of hanging a cm off the bed, with 3/4 of a cm of garbled mess at the bottom, and then a perfectly printed support structure above it.

I've also had interesting results printing with 15% flow.  A benchy built on top a 15%-flow block came out good, but with a hollow bottom.  The first layer has to be 100% flow or else it won't adhere.

This saves on material.  potential approaches include building 1mm-wide solid pillars or otherwise as scaffold, throwing a few layers of stronger support at the top, abusing the fact that several layers of drooping overhangs give support to layers above them, and so forth.

Do note I used 25% gyroid infill, and issued an M221 S15 after layer 1, and an M221 S100 at the first layer where only the benchy outline was printed.  I also did this on a 0.6mm nozzle.  If you want to get artistic with it, by minimizing walls, using somewhat sparse infill, and a low flow rate, you can play around with sticking miniature benchies at various orientations around in 3D space and see if they stick.  I haven't found a way to make a block in PrusaSlicer that just does this; it's one way to get an idea of what you'd need for really sparse supports.  The fact that the print recovers from bad overhang layers after several further layers gives a lot of options to save filament.  Most supports do not rely on strength.

_Is this a new feature request?_  Yes

rtyr, 2021-02-21:
So basically you are requesting flow setting for the supports? If you need it just for some experiments, you can use virtual extruder for it (increase number of extruders, then set different filament profile for the support - profile with different multiplier).

rtyr, 2021-06-17:
No response. Closing.

closed 2021-06-17 by rtyr