prusa3d/PrusaSlicer #6092: Feature request:  Autotemp

jrmoserbaltimore, 2021-02-20:
### Version
2.3.0

### 3D printer brand / version + firmware version (if known)
marlin 2.0.7

### Behavior

_Is this a new feature request?_ Yes.

In Filament Settings, under Temperature, it would be nice to set "other layers" by a minimum-maximum, issued by `M104 F{Factor} S{Minimum Temperature} B{Maximum Temperature}`.  `{Factor}` can be omitted to use the default set by firmware.

Perhaps something like:

- First layer temperature
- Other layers temperature
- Automatic temperature control (Checkbox)
    - Maximum temperature
    - Factor (0 for firmware default)

It is perhaps wise to set a maximum temperature in Extruder under Printer Settings as well.

rtyr, 2021-02-21:
It is supported only by Marlin FW as far as I know. I'm not sure, how many people actually use it.

marcoa131, 2022-11-20:
> It is supported only by Marlin FW as far as I know. I'm not sure, how many people actually use it.

I do use it on some of my machines that run Marlin, it allows me to quickly print strong parts with good overhangs and a lot of detail by effectively going past the max volumetric flow rate of the hotend for the base temperature of the print.
that spares me from having to change the hotend of the printer for a high-flow one and keeping the ability of printing small, and delicate parts on the same machine.

It would be very interesting to see such feature implemented at the level of the slicer, that way anyone could use this feature on any printer without having to run user-compiled Marlin on their boards.

closed 2026-09-29 by prusaslicer-bot[bot]