prusa3d/PrusaSlicer #6093: under-extrusion artifacts on perimeters after tiny angled moves - linear advance artifact TBC

Moyrn, 2021-02-20:
### Version
Build:     PrusaSlicer-2.3.0+win64-202101111315

### Operating system type + version
Windows Version:     Windows 10 (build 18363), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Prusa  i3 MK3S/MMU2S - firmware 3.9.3

### Behavior
under-extrusion artifacts on perimeters after tiny angled moves - linear advance artifact TBC

Context :
- I use small prints to display information (serial number, range, brand...). These are engraved (not embossed)
- Nozzle 0.4 requires to force external perimeters to 0.25 to display small text properly
- All is perfectly calibrated (X,Y,Z,E, No frame skew, lin advance K0.05)
- Perimeters 25mms/s - 100mm/s² - Jerk 8/8
- setup is on a heavy furniture (no wobbling at all)
- Printer tweaked with Misumi rods & bearings, BNBSX extruder, updated steppers (X/Y Trinamic QSH4218-51-10-049, Z Stepperonline 17HS10-0704S), MMU2S with new blade + magnet holder 
- Filament NOULEI SILK (copper, silver, ...). So shiny that retractation is not used (leaves scars anytime). Also quite "soft". The standard extruder brought the filament too hot with the direct motor-shaft > constant grinding). Now all OK with BNBSX since one year. 

---- Now hunting progressively on "print quality"

Problem :
- After having done these small texts perimeters, some under extrusion happen
- Please note some ghosting also appears (never could remove it)
- Appears after each text, on any side of the print (there are also on the back - same behaviour)

---- all is perfect except around these text prints ----

Assumption :
- Speed decrease when doing tiny text perimeters
- Speed increase when accelerating again to print the linear shape
- The acceleration is seemingtly "braked"
- There seems to be a conflict with linear advance - something prevents to go full speed
- Rings a bell as I found something related in a presumably fixed issue (from 3.9.1)
> Major improvements to LA15
> The Linear Advance 1.5 (LA15) functionality was first introduced in the firmware release 3.9.0 and brought several improvements both to the print speed and quality. However, in certain scenarios, the algorithm didn't perform as expected during deceleration, leaving artifacts on specific geometries.
> 
> The issue was discovered in a bad timer check, which prevented fast LA ticks to be scheduled, effectively limiting LA as speed increased (such as in long stretch of perimeters or during infill) or with shorter intervals due to geared extruders (commit 173aa2d). This also combined with a bad sign check that would cause over-retraction as extrusion speed/nozzle size increased (commits c08f37d & fb5f09d).)

Expections / help expected 
- Smooth layers as with areas with "no text"
- Any idea on the ghosting effect before/after letters ? i already print very slow... but i'm sure a miss a trvial thing. thanks for your kind attention.

Tests done: 
- From K0 to K 0.15
- Jerk 2/2

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[image]
[image]

3MF file is a simplified version to "focus" on the issue only
[TEST-PERIMETERS-BUGREPORT.zip](https://github.com/prusa3d/PrusaSlicer/files/6014495/TEST-PERIMETERS-BUGREPORT.zip)

rtyr, 2021-02-21:
It seems more like issue for prusa3d/Prusa-Firmware, don't you think?

Moyrn, 2021-02-21:
Hi/ Thanks for taking the time to read and reply.
I played for 2 days on this matter.

There is something weird. I dropped "short perimeters speed" below (like : 10 or 15). Even played with 0.
Back to K=0.05 or 0.03 for Okayish results. I got rid of the under-extrusion. But now I have inconsistency bands. The print is larger where there is no deceleration (no letter). See below
[image]

Now i tried mechanical fixes (x, y, clean-up, oil, belts...). No change.
I got bored and I tried on Cura 4.6 which still holds a similar profile for me (K0.05, etc...). 
Damn ! Perfect ! So not the mechanical nor the firmware. 
See below.
[image]

Moyrn, 2021-02-22:
Made a full print (not a sample) on Cura, and finally I also get some identical artifacts as on Slicer. So you are possibly right with firmware assumption/

Moyrn, 2021-02-22:
Checked firmware / rebuilt-it / re-flashed. No change

Moyrn, 2021-02-23:
I started from fresh with std filament & print settings. Some progress. 
The only thing which is not "gained" is the stripes (as on picture above the one above) : less present, but still there.
BUT NEVER on CURA. Any idea where these would come from ?

Moyrn, 2021-02-23:
Got some progress again. 
- with high K, loose jerk
- short perimeters on carved letters creates rebounds > slower has better results
- Using a width of 0.25 for a 0.4 nozzle on external perimters somehow messes with linear advance. I can't get it to work on all cases. It goes fine with 0.4... but fine embeded/carved prints do not render well. I can try to play with flow, filament size, temp... nothing is reliable enough.

While on Cura, even a 0.23w (nozzle 0.4) is OK. But Cura is not straighforward with MMU2S, has a poorer ironing...

I will still try a couple of features... Talk to you later...

rtyr, 2021-02-23:
I wouldn't expect good/reliable results with such extreme settings. I think the best solution here is to use smaller diameter nozzle.

Moyrn, 2021-02-23:
Yes. I used to run the .25 nozzle. Damn long prints, and more issues with MMU2S (filament tips, unload/load...).
So in a way, i got more reliable with .4 (less fails), but less clean perimeters for sure. But something changed recently. Can't say yet. Cura .23 is extreme but only for "tests". not for real cases. Though the print you see above looks ok with this. 
Thanks for sharing.

Moyrn, 2021-02-27:
Got some progress. Got rid of the underextrusion signs & of the thickness stripes. Even on Prusa Slicer. Only ghosting i could not remove. Too slow or High K or Low Jerk > surface is poor, no ghosting. Some choice to be done... maybe i'll try again later. But spending about 8-9 days on it is already a pain...

Want to share this in case someone falls in the same cases. These are possible escapes.

My stable parameters
- Noulei silk filament K0.05 1.73mm 105% - 205°/60°C - Fan 75%
- Jerk 8/8
- perimeters acceleration 300
- short perimeters speed 10
- ext perimeters speed 25
- perimeters width 0.35 (to stick of to my design... will increase later)
- external perimeters 0.3 (instead of 0.25 for more consistency, my letters remain readible)

Uncertain
- Resolution 0.05 > wanted to check if small artifacts were provoking small moves. Can't see difference on these designs

I'm still in need for some help about ghosting if some of you are still reading...

closed 2026-09-29 by prusaslicer-bot[bot]