prusa3d/PrusaSlicer #6094: Slicer crashes in MMU2S mode

Ikif, 2021-02-20:
### Version
2.3.0 + win64

### Operating system type + version
win10

### 3D printer brand / version + firmware version (if known)
MK3S + MMU2S

### Behavior
I define MK3S & MMU2S.
Than I change to 3 extruders , and after I press right mouse button  on model , the slicer every time crashed

lukasmatena, 2021-02-20:
Please post a 3MF.

Ikif, 2021-02-21:
[amongusBody.zip](https://github.com/prusa3d/PrusaSlicer/files/6016914/amongusBody.zip)

bubnikv, 2021-02-22:
It crashes in void ObjectList::append_menu_item_change_extruder(wxMenu* menu)
in
*icons[icon_idx]
where icons is 4 items long and icon_idx = 4
@YuSanka please look into the issue. Thanks.

Ikif, 2021-02-22:
> It crashes in void ObjectList::append_menu_item_change_extruder(wxMenu* menu)
> in
> *icons[icon_idx]
> where icons is 4 items long and icon_idx = 4
> @YuSanka please look into the issue. Thanks.

Sorry, that doesn't say me too much.

YuSanka, 2021-02-22:
> > It crashes in void ObjectList::append_menu_item_change_extruder(wxMenu* menu)
> > in
> > *icons[icon_idx]
> > where icons is 4 items long and icon_idx = 4
> > @YuSanka please look into the issue. Thanks.
> 
> Sorry, that doesn't say me too much.

@Ikif , It was a message for me :)

YuSanka, 2021-02-22:
Thanks for your report. This bug is fixed and will be a part of next release.

closed 2021-02-22 by YuSanka