prusa3d/PrusaSlicer #6095: Preview Tool-View is broken

JoeMudr, 2021-02-20:
### Version
Version 2.3.0+linux-x64

### Operating system type + version
Ubuntu 20.10

### 3D printer brand / version + firmware version (if known)
Anycubic i3 Mega 1.1.2

### Behavior
Preview Tool-View is broken. 
Select layer one. It shows travels for all layers but no lines.
[image]

rtyr, 2021-02-21:
Please provide your 3MF project file.

JoeMudr, 2021-02-21:
Happens basically with every model i tried the last days but in slightly different variations.
[Retract(1).zip](https://github.com/prusa3d/PrusaSlicer/files/6018492/Retract.1.zip)

rtyr, 2021-02-21:
I was able to reproduce it. It seems it happens only when the view is set to "tool"+enabled Travel option.

rtyr, 2021-04-12:
Fixed with https://github.com/prusa3d/PrusaSlicer/commit/613f961b9fab7fb5515644d49585fd3447079e5a. Closing.

closed 2021-04-12 by rtyr

bubnikv, 2021-04-14:
Will be fixed in PrusaSlicer 2.3.1-rc