prusa3d/PrusaSlicer #6096: PrusaSlicer instability + crashes

jbro-oregon, 2021-02-20:
### Version
2.3.0+win64

### Operating system type + version
Windows 7 Professional 64-bit

### 3D printer brand / version + firmware version (if known)
Prusa i3 mk2s, highly modified, based on Bear i3

### Behavior
After several iterations of modifying PrusaSlicer settings and re-slicing, the software responsiveness takes a hit. Eventually the software just crashes and must be terminated. Some of the screen information becomes garbled (see picture), which is a sure sign that it is about to crash.
I thought that this was isolated when I first installed 2.3.0, but it happens pretty regularly.
Debug info: "Unhandled exception at 0x000007FEB1322EFA (PrusaSlicer.dll) in prusa-slicer.exe: 0xC000041D: An unhandled exception was encountered during a user callback."
My PC is very capable (designed for SolidWorks and Photoshop). I have 32GB of RAM, NVME SSD and an NVIDIA Quadro P2200 GPU. I do not have stability issues with any other installed software, or with Windows itself.

[image]

jbro-oregon, 2021-02-20:
I forgot to mention that the file being sliced was opened from a local NAS drive... This shouldn't matter, but it is different than normal for most users.

bubnikv, 2021-02-22:
Thanks for report, but unfortunately your report is not sufficient to
reproduce your issue. We need more details./

lukasmatena, 2024-02-18:
We never reproduced this and I don't remember anyone else reporting this. 
Closing as outdated.

closed 2024-02-18 by lukasmatena