prusa3d/PrusaSlicer #6097: Extruder Selection Does Not Respond

Darkknight33, 2021-02-21:
### Version
2.3

### Operating system type + version
MacOS Big Sur

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S+ MMU2S

### Behavior
Load-in Multi Material Object
Slicer recognizes multi part object, click okay.
Changing any extruder assignment in body tree is completely impossible. Does not respond to any amount of left/right clicking or keyboard actions.
Same process on Window 10 works as expected, can assign any extruder to any part.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
https://www.thingiverse.com/thing:2649487

FidelCapo, 2021-02-22:
Unfortunately, I am not able to reproduce this issue. Right click on the color rectangle works properly for me on intel Macbook pro 2017 and M1 Macbook air 2020. Do you have external monitors connected or custom resolution/scaling set?

Darkknight33, 2021-02-22:
No monitors or any external hardware. It’s just the laptop. 

After just investigating this a bit more on my end, the problem was in fact between the chair and the keyboard. I am new to Mac and did not realize you need to control-click or ‘two-finger’ click the trackpad to achieve the equivalent of a right click. Not exactly my proudest moment.

Still, seems like there is no reason to require a secondary click to open the extruder assignment menu, when primary click does nothing. On Windows, it’s a single left click to open the same menu. Is there a design reason for it having different behavior on Mac?

bubnikv, 2021-02-23:
> Still, seems like there is no reason to require a secondary click to open the extruder assignment menu, when primary click does nothing. On Windows, it’s a single left click to open the same menu. Is there a design reason for it having different behavior on Mac?

It is a quirk of wxWidgets, the user interface toolkit.

closed 2021-02-25 by Darkknight33