prusa3d/PrusaSlicer #6098: Feature Request: Preview print temperatures in Preview

LightningShark25, 2021-02-21:
### Version
2.3.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
All - currently Ender 3 v2

### Behavior
New Feature Request

In Preview you can select a number of attributes to preview by layer/colour e.g.

[image]

It would be awesome if temperature could be added to this list please?

As you can see from my example I'm working on a temperature tower.
It would be handy to be able to see the temperature changes in preview (I'm using custom gcode to modify temps) without having to dig through the gcode output.

Thanks in advance :)

LightningShark25, 2021-02-21:
I'm new to GitHub, couldn't work out how to tag this as Feature Request sorry!

rtyr, 2021-02-21:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4402. Closing.

closed 2021-02-21 by rtyr

combolek, 2021-02-22:
FWIW, see PR #6106