prusa3d/PrusaSlicer #6099: STL which doesn't slice properly

dheera, 2021-02-21:
Attached STL doesn't slice properly

[problem.stl.gz](https://github.com/prusa3d/PrusaSlicer/files/6017280/problem.stl.gz)

### Version
2.3.0

### Operating system type + version
Ubuntu 20.04

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3

### Behavior
[image]
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

dheera, 2021-02-21:
Another one:

[problem2.stl.gz](https://github.com/prusa3d/PrusaSlicer/files/6017293/problem2.stl.gz)

[image]
[image]

dheera, 2021-02-21:
Also, Slicer3 seems to handle these STLs just fine.

rtyr, 2021-02-21:
This problem is already fixed by https://github.com/prusa3d/PrusaSlicer/commit/9c5dd358ab23e9351dce0b9c71eaa5ddf92d6f17. The fix will be included in the next PrusaSlicer release. Both STLs (generated in OpenSCAD) are sliced fine with current master.

Commit info:

> Fixing model import of models with nearly degenerate triangles, which
> degenerate after transformation (translation or rotation).
> 
> There was certainly a bug in PrusaSlicer, however this bug was triggered
> mostly by the ASCII STL files generated by OpenSCAD, likely due to insufficient
> number of decimal digits when saving an ASCII STL, see
> [openscad/openscad] STL export should contain single precision floats (#2651)

Closing.

closed 2021-02-21 by rtyr