prusa3d/PrusaSlicer #6100: Cannot add new support points on some part of SLA prints

Blup1980, 2021-02-21:
### Version
Version 2.3.0+win64

### Operating system type + version
PrusaSlicer
Version:   2.3.0+win64
Build:     PrusaSlicer-2.3.0+win64-202101111315

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 34,268MB
OpenGL installation
GL version:   4.6.0 NVIDIA 461.40
Vendor:       NVIDIA Corporation
Renderer:     GeForce RTX 2080 Ti/PCIe/SSE2
GLSL version: 4.60 NVIDIA

### 3D printer brand / version + firmware version (if known)
Prusa SL1

### Behavior
I cannot add supports on some part of the sliced object

Step 1:
(figure A) Open Prusa Slicer
[image]

Step 2:
(figure B) Load project DeReport.3mf
[image]

Step 3:
(figure C) Select the part and click "manual editing" of the support points
[image]

Step 4:
(figure D) Left click anywhere on the face shown into the red area of the figure
[image]

* Expected Results:
A left click should add another support point. This is the case on some other area of the same sliced object

* Actual Results:
A left click on that area does nothing at all.

#### Project File (.3MF) where problem occurs

[DeReport.zip](https://github.com/prusa3d/PrusaSlicer/files/6017374/DeReport.zip)

msspad, 2021-08-22:
This is a very annoying bug. The workaround is creating supports in Chitubox, transfer the stl file to PrusaSlicer and continue.

lukasmatena, 2021-09-24:
Should be fixed by 6b25a9c for PrusaSlicer 2.4.0-alpha2. Please, retest after it is released. Thanks.

lukasmatena, 2021-09-24:
Sorry, correction. The fix is buggy, it will not be part of 2.4.0-alpha2.

commit 6b25a9c8 (2021-09-24): Fixed volume transformations in SLA:
Volume transformations were ignored in SLA mode. This did not matter for plain STLs
and PS own 3MF, because in those cases, the volume trafo was identity. Importing
a 3rd party 3MF leads to issues with support/holes placement and generation.
Fixes #6100 and #6744.