prusa3d/PrusaSlicer #6103: Weird pattern on the first layer

JoeMudr, 2021-02-21:
### Version
Version 2.3.0+linux-x64

### Operating system type + version
Ubuntu 20.10

### 3D printer brand / version + firmware version (if known)
Anycubic i3 Mega 1.1.2

### Behavior
Weird pattern on the first layer.
Default extrusion width: 0.4mm
First layer: 0.45mm
In some areas it seems to print the regular width of 0.4mm.
I can actually see this while printing. The smooth lines squeeze a bit around the nozzle. The messed up lines are perfectly flat on the top.

Same settings in cura (left)are perfectly fine.
[image]

rtyr, 2021-02-21:
Use "Monotonic" bottom/top fill pattern instead of "Rectiliniear", it will produce more uniform surface. Otherwise it is about first layer calibration and leveling.

[image]

JoeMudr, 2021-02-21:
Perfect! Thank you!
[image]

rtyr, 2021-02-22:
Solved. Closing.

closed 2021-02-22 by rtyr