prusa3d/PrusaSlicer #6104: Prusa slicer 2.3.0 cannot slice using hilbert curve top pattern when it is a bigger print surface

Alucard-i, 2021-02-21:
### Version
_PrusaSlicer-2.3.0+win64-202101111315_

### Operating system type + version
_Win10 #s_

When slicing bigger surfaces using hilbert curve i have issues. In old version before 2.3.0 i sliced my models with hilbert curve in about 15 seconds now i can leave them for 30 minutes and they will not be done. I have a  Ryzen 9 3900 so hardvare are not the issue. I cannot share my files cause it is files i use and sell the models from.
I have Tried my saved 3mf files and i have tried now STL files, it makes no differens.

rtyr, 2021-02-21:
Same as https://github.com/prusa3d/PrusaSlicer/issues/5771. It was already fixed for next PrusaSlicer release. Closing.

closed 2021-02-21 by rtyr

Alucard-i, 2021-04-21:
Well I just tried 2.3.1 and still same problems.

rtyr, 2021-04-21:
I meant 2.4.0 (alpha) by next PrusaSlicer release. That fix is not included in 2.3.1. Differences between 2.3.0 and patch release 2.3.1 are mentioned here https://github.com/prusa3d/PrusaSlicer/releases.

Alucard-i, 2021-04-21:
> I meant 2.4.0 (alpha) by next PrusaSlicer release. That fix is not included in 2.3.1. Differences between 2.3.0 and patch release 2.3.1 are mentioned here https://github.com/prusa3d/PrusaSlicer/releases.

So I will be stuck with 2.2.x until 2.4.x that is kinda shitty.

rtyr, 2021-07-02:
@Alucard-i 

https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.2-beta