prusa3d/PrusaSlicer #6105: Crashes while reading config

dsvi, 2021-02-21:
### 2.3.0
### Suse Tumbleweed

The config attached:

[~/.config/PrusaSlicer](https://github.com/prusa3d/PrusaSlicer/files/6018840/PrusaSlicer.zip)

combolek, 2021-02-22:
FWIW, I cannot reproduce it
[image]

bubnikv, 2021-02-22:
I cannot reproduce the crash either, with ASAN build on Ubuntu 20.

dsvi, 2021-02-22:
Weird. For me, it says this:

```
02:41:47 AM: Debug: window wxTreeCtrl(0x55fe3f9d0510) lost focus even though it didn't have it
02:41:47 AM: Debug: window wxTreeCtrl(0x55fe3f9d0510) lost focus even though it didn't have it
02:41:47 AM: Debug: window wxTreeCtrl(0x55fe3f9d0510) lost focus even though it didn't have it
02:41:47 AM: Debug: window wxTreeCtrl(0x55fe3f9d0510) lost focus even though it didn't have it
02:41:47 AM: Debug: window wxTreeCtrl(0x55fe3f9d0510) lost focus even though it didn't have it
02:41:47 AM: Debug: window wxTreeCtrl(0x55fe3f9d0510) lost focus even though it didn't have it
02:41:47 AM: Debug: window wxTreeCtrl(0x55fe3f9d0510) lost focus even though it didn't have it
02:41:47 AM: Debug: window wxTreeCtrl(0x55fe3f9d0510) lost focus even though it didn't have it
02:41:47 AM: Debug: window wxTreeCtrl(0x55fe3f9d0510) lost focus even though it didn't have it
02:41:47 AM: Debug: window wxTreeCtrl(0x55fe3f9d0510) lost focus even though it didn't have it
[2021-02-22 02:41:47.309567] [0x00007f3ca4e988c0] [error]   Uncaught exception: basic_string::substr: __pos (which is 38) > this->size() (which is 34)
Unhandled standard exception of type "St12out_of_range" with message "basic_string::substr: __pos (which is 38) > this->size() (which is 34)"; terminating the application.
[2021-02-22 02:41:47.403522] [0x00007f3ca4e988c0] [error]   Uncaught exception: basic_string::substr: __pos (which is 38) > this->size() (which is 34)
[2021-02-22 02:41:47.417854] [0x00007f3ca4e988c0] [error]   Uncaught exception: basic_string::substr: __pos (which is 38) > this->size() (which is 34)
Unhandled standard exception of type "St12out_of_range" with message "basic_string::substr: __pos (which is 38) > this->size() (which is 34)"; terminating the application.
Aborted (core dumped)
```
Which WxWidgets are you using? 3.0 or 3.1?

bubnikv, 2021-02-22:
> Which WxWidgets are you using? 3.0 or 3.1?

patched 3.1.4, compiled with our deps script

dsvi, 2021-02-24:
Apparently the Appimage doesn't have the issue. Thnx!

closed 2021-02-24 by dsvi