prusa3d/PrusaSlicer #6107: Feature Request: Hide bed texture/logo when viewing through/under toward the part

LightningShark25, 2021-02-22:
### Version
2.3

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 3 v2 Marlin 2.x

### Behavior
The print bed logo 'Ender' (perhaps this is specific to my build of PrusaSlicer) appears on the bed in the middle.
This is not a user definable texture - it just seems to appear built in.

When you orbit the view so you look under the model (through the build plate), the logo gets in the way of viewing the part.
[image]

I suggest the logo is hidden when viewing through the bed or atleast made very transparent.  Or add an option to hide the texture?

For those also having this issue, one workaround is to define the bed 'texture' in Printer Settings > Bed Size, to a blank grey image created in paint.

Thanks for your time

foreachthing, 2021-02-22:
Not for me, thank you (or, please make it optional).
I got a grid and I'd still like to see it, even when viewed from the bottom.

@LightningShark25, why don't you move the logo to a (seldom used) corner?
Or even "better": make the logo more transparent (50%, instead of 0).

Not perfect, but a bit better:
[image]

LightningShark25, 2021-02-22:
Configurable is fine...

When using a user defined 'texture' the current behavior is the textured bed plate goes transparent - but still visible, which I like.  I don't know why the 'default' Ender logo (which I didn't configure) doesn't also go transparent.  

I personally don't want to see the logo at all :)

foreachthing, 2021-02-22:
There's a svg file you can edit in inkscape. Then make the layer with the logo transparent.

LightningShark25, 2021-02-26:
@foreachthing   thanks for the tip - I'll search for it

foreachthing, 2021-02-28:
The svg in question, is located under: 
PrusaSlicer/resources/profiles/Creality

LightningShark25, 2021-02-28:
Thanks. I haven't had time to test this out myself yet. But I plan to.

The question is, would it be of benefit to the community to add a configurable  option for  transparency when viewing below as a standard feature? Low priority yes as there is a workaround but still....

thijstriemstra, 2022-03-19:
The ability to hide the build plate is something that I couldn't find, and I stumbled upon this ticket. Is it really not possible to hide these items (texture/logo/outline of buildplate)?

foreachthing, 2023-04-14:
Well, you could just load an empty .png/.svg:
[image]
A .png needs to be only 1x1 pixel and can be transparent. Or, use a search engine to find an empty image.

psi-4ward, 2023-04-14:
Nice idea but the shape gets white than

r4v5, 2023-07-21:
The image of the build plate is indeed removable for folks who don't like it, it's the 'Remove' button of the image in the screenshot above. You get a grid instead:
[image]

That said, a build plate texture _is_ also rendered with transparency when viewing from below, at least on 2.6:
[image]
(ignore the z-fighting on the texture when viewed from above, that's a separate issue)

thijstriemstra, 2023-07-29:
> Well, you could just load an empty .png/.svg:

That doesn't hide the build plate, only the logo/text stuff?

closed 2026-09-29 by prusaslicer-bot[bot]