prusa3d/PrusaSlicer #6109: Seam enforcers/blockers are lost after reload from disk

adamdport, 2021-02-22:
### Version
_Version of PrusaSlicer used goes here_
2.3.0+

### Operating system type + version
Mac 11.2.1

### Behavior
If you apply seam enforcers and then do "reload from disk", the seam enforcers are lost, even if you reload the exact same STL.

_Is this a new feature request?_
Maybe? I'm assuming reload from disk was intended to support this?

lukasmatena, 2021-02-22:
Thanks, this is already fixed in master (d001195ebd2239ba21f31a908d1c146e7af03c08). It will work in the next release. Closing.

closed 2021-02-22 by lukasmatena

lukasmatena, 2021-08-05:
I just reverted the "fix" with 413dc8d6ec1630573ab159cef899599fd755816f. Throwing away the paint-on data on reload from disk was correct. The reason one uses reload from disk is that the file has changed, so the paint-on regions might be recreated incorrectly or (worse) make slicer crash. I was too fast before.