prusa3d/PrusaSlicer #6114: Printer and Feature request

SPCORRIV, 2021-02-23:
Pursa Slicer 2.3.0

OS MAC Big Sur

Please Add Creality CR-6 SE into profiles.

Issue 
For some reason some print materials are blocked per printer please unlock all filaments for printers as a lot of us have modified our printers to except optional filaments like TPU and nylons

Thank You

rtyr, 2021-02-23:
Creality CR-6 SE profile is already prepared for next PrusaSlicer release.

Filament profiles are bound to vendor space. It makes sense, because filament profiles are often not compatible with different printers (different temps, retraction overrides, linear advance, etc.). But we may introduce some set of "generic material profiles".

Closing.

closed 2021-02-23 by rtyr

SPCORRIV, 2021-02-23:
Thank you very much any idea when the update will be released  ?

erickwill, 2021-02-23:
> Creality CR-6 SE profile is already prepared for next PrusaSlicer release.
> 
> Filament profiles are bound to vendor space. It makes sense, because filament profiles are often not compatible with different printers (different temps, retraction overrides, linear advance, etc.). But we may introduce some set of "generic material profiles". There are already opened issues on this topic.
> 
> Closing.

Such awesome new! May I suggest a weekly branch with latest updates? It would be great to have the cutting edge merges weekly.