prusa3d/PrusaSlicer #6118: non-needed support structures

wvcollenburg, 2021-02-23:
### Version
Version 2.3.0+win64
PrusaSlicer-2.3.0+win64-202101111315

### Operating system type + version
PrusaSlicer
Version:   2.3.0+win64
Build:     PrusaSlicer-2.3.0+win64-202101111315

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 34,291MB
OpenGL installation
GL version:   4.6.0 NVIDIA 460.79
Vendor:       NVIDIA Corporation
Renderer:     GeForce RTX 2080/PCIe/SSE2
GLSL version: 4.60 NVIDIA

### 3D printer brand / version + firmware version (if known)
Prusa i3MK3s + MMU2s (filament profile has a modified ramming setting. other than that it is the default prusament profile.

### Behavior
When slicing a model with painted support and a support modifier it generates a support structure around the object that serves no apparent purpose and seems a waste of time and filament.

#### Project File (.3MF) where problem occurs

[outercase.zip](https://github.com/prusa3d/PrusaSlicer/files/6031811/outercase.zip)

wvcollenburg, 2021-02-23:
This is not a report that prevents anything from working properly... just wanted to bring this to your attention.

bubnikv, 2021-10-05:
[image]

closed 2026-09-29 by prusaslicer-bot[bot]