prusa3d/PrusaSlicer #6126: PlaceholderParser doesn't suffix zeros

pmjdebruijn, 2021-02-24:
### Version
`2.3.0~336+g6bdd42b1c`

### Behavior
`output_filename_format = whatever_{layer_height}_whatever.gcode`

Produces an improper file sort order, due to a lack of suffixed zeros:

``
whatever_0.08mm_whatever.gcode
whatever_0.12mm_whatever.gcode
whatever_0.16mm_whatever.gcode
whatever_0.1mm_whatever.gcode
whatever_0.24mm_whatever.gcode
whatever_0.28mm_whatever.gcode
whatever_0.2mm_whatever.gcode
``

It should be

``
whatever_0.08mm_whatever.gcode
whatever_0.10mm_whatever.gcode
whatever_0.12mm_whatever.gcode
whatever_0.16mm_whatever.gcode
whatever_0.20mm_whatever.gcode
whatever_0.24mm_whatever.gcode
whatever_0.28mm_whatever.gcode
``

I think I've found relevant bits of code:

https://github.com/prusa3d/PrusaSlicer/blob/34417574f5a823994b0e16beb97d2bac877ecf6f/src/libslic3r/PlaceholderParser.cpp#L257

And:

https://github.com/prusa3d/PrusaSlicer/blob/34417574f5a823994b0e16beb97d2bac877ecf6f/src/libslic3r/PrintConfig.cpp#L79

Presumably that should be `ConfigOptionFloat(0.2)`

I've tried changing this locally, but it does trip up some of the tests:

``
The following tests FAILED:
	  2 - libslic3r_tests (Failed)
	  4 - fff_print_tests (Failed)
``

bubnikv, 2021-02-24:
There is nothing wrong with the PlaceholderParser. What you want is a new
feature - a number formatting facility inside the placeholder parser.

bubnikv, 2021-02-24:
https://github.com/prusa3d/PrusaSlicer/issues?q=is%3Aopen+is%3Aissue+label%3A%22placeholder+parser%22

Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/1680#issuecomment-737702130

closed 2021-02-24 by bubnikv