prusa3d/PrusaSlicer #6134: Application exits while adjusting range of Height range modifier

bumerankkk, 2021-02-25:
### Version
2.3.0

### Operating system type + version
Linux Elementary OS 5.1.7 Hera (based on Ubuntu 18.04.4 LTS) / Linux 5.3.0-62-generic

### 3D printer brand / version + firmware version (if known)
does not matter

### Behavior
Application exits without notice when these steps done:

1. Enable Height range modifier on the object
2. Type in new value of "start at height" or "stop at height"
3. press Enter

I started PrusaSlicer from CLI and here is the complete console output:
```$ prusaslicer 

(prusa-slicer:21145): Gtk-WARNING **: 20:37:26.405: Unable to locate theme engine in module_path: "pixmap",

(prusa-slicer:21145): Gtk-WARNING **: 20:37:26.406: Unable to locate theme engine in module_path: "pixmap",

(prusa-slicer:21145): Gtk-WARNING **: 20:37:26.406: Unable to locate theme engine in module_path: "pixmap",

(prusa-slicer:21145): Gtk-WARNING **: 20:37:26.406: Unable to locate theme engine in module_path: "pixmap",
Gtk-Message: 20:37:26.406: Failed to load module "canberra-gtk-module"

(prusa-slicer:21145): GLib-GObject-WARNING **: 20:37:26.611: invalid (NULL) pointer instance

(prusa-slicer:21145): GLib-GObject-CRITICAL **: 20:37:26.611: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:21145): GLib-GObject-WARNING **: 20:37:26.611: invalid (NULL) pointer instance

(prusa-slicer:21145): GLib-GObject-CRITICAL **: 20:37:26.611: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:21145): GLib-GObject-WARNING **: 20:37:26.611: invalid (NULL) pointer instance

(prusa-slicer:21145): GLib-GObject-CRITICAL **: 20:37:26.611: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:21145): GLib-GObject-WARNING **: 20:37:26.611: invalid (NULL) pointer instance

(prusa-slicer:21145): GLib-GObject-CRITICAL **: 20:37:26.611: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:21145): GLib-GObject-WARNING **: 20:37:26.738: invalid (NULL) pointer instance

(prusa-slicer:21145): GLib-GObject-CRITICAL **: 20:37:26.738: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:21145): GLib-GObject-WARNING **: 20:37:26.738: invalid (NULL) pointer instance

(prusa-slicer:21145): GLib-GObject-CRITICAL **: 20:37:26.738: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:21145): GLib-GObject-WARNING **: 20:37:26.739: invalid (NULL) pointer instance

(prusa-slicer:21145): GLib-GObject-CRITICAL **: 20:37:26.739: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:21145): GLib-GObject-WARNING **: 20:37:26.739: invalid (NULL) pointer instance

(prusa-slicer:21145): GLib-GObject-CRITICAL **: 20:37:26.739: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:21145): GLib-GObject-WARNING **: 20:37:26.739: invalid (NULL) pointer instance

(prusa-slicer:21145): GLib-GObject-CRITICAL **: 20:37:26.739: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:21145): GLib-GObject-WARNING **: 20:37:26.739: invalid (NULL) pointer instance

(prusa-slicer:21145): GLib-GObject-CRITICAL **: 20:37:26.739: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:21145): GLib-GObject-WARNING **: 20:37:26.740: invalid (NULL) pointer instance

[... 85 lines cut ...]

(prusa-slicer:21145): GLib-GObject-WARNING **: 20:37:26.753: invalid (NULL) pointer instance

(prusa-slicer:21145): GLib-GObject-CRITICAL **: 20:37:26.753: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:21145): GLib-GObject-WARNING **: 20:37:26.753: invalid (NULL) pointer instance

(prusa-slicer:21145): GLib-GObject-CRITICAL **: 20:37:26.753: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:21145): GLib-GObject-WARNING **: 20:37:26.755: invalid (NULL) pointer instance

(prusa-slicer:21145): GLib-GObject-CRITICAL **: 20:37:26.755: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:21145): GLib-GObject-WARNING **: 20:37:26.755: invalid (NULL) pointer instance

(prusa-slicer:21145): GLib-GObject-CRITICAL **: 20:37:26.755: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
malloc_consolidate(): invalid chunk size
Neúspěšně ukončen (SIGABRT)
```

_Is this a new feature request?_
Nope

#### Project File (.3MF) where problem occurs
Any

lukasmatena, 2021-02-25:
Thanks for the report, but I can't reproduce it on Ubuntu 16.04. Could you please list all steps that you do since the application starts, share your 3MF and ideally provide a video with the shortest way to reproduce the crash? It may well be caused by something innocent you do before pressing the Enter. Does it happen with our official AppImage build or did you compile it yourself?

lukasmatena, 2021-02-26:
Now I tried in Debug mode (I was running Release + ASAN before) and it indeed crashes even on my system. The backtrace looks interesting. I saw it crash with two different backtraces, actually:

<details>
<summary>Click to see backtrace 1</summary>
<br>

1  malloc_consolidate                                                                                                                                                                                                                                                                                                              malloc.c             4176 0x7ffff457846b 
2  _int_malloc                                                                                                                                                                                                                                                                                                                     malloc.c             3457 0x7ffff457bd0c 
3  _int_realloc                                                                                                                                                                                                                                                                                                                    malloc.c             4311 0x7ffff457d2e0 
4  __GI___libc_realloc                                                                                                                                                                                                                                                                                                             malloc.c             3052 0x7ffff457e8a9 
5  g_realloc                                                                                                                                                                                                                                                                                                                                                 0x7ffff6af7878 
6  ??                                                                                                                                                                                                                                                                                                                                                        0x7ffff6ac5399 
7  g_array_insert_vals                                                                                                                                                                                                                                                                                                                                       0x7ffff6ac5855 
8  ??                                                                                                                                                                                                                                                                                                                                                        0x7ffff53b5aa4 
9  ??                                                                                                                                                                                                                                                                                                                                                        0x7ffff53b5cea 
10 ??                                                                                                                                                                                                                                                                                                                                                        0x7fffedfbb85c 
11 ??                                                                                                                                                                                                                                                                                                                                                        0x7ffff53b7962 
12 gtk_rc_get_style                                                                                                                                                                                                                                                                                                                                          0x7ffff53bb8c6 
13 ??                                                                                                                                                                                                                                                                                                                                                        0x7ffff548e9b0 
14 wxWindow::GTKApplyStyle                                                                                                                                                                                                                                                                                                         window.cpp           5576 0x22c277f      
[... cut ...]
90 main                                                                                                                                                                                                                                                                                                                            PrusaSlicer.cpp      798  0x47a53b       
</details>

My wild guess is a double-free somewhere (maybe in `Slic3r::GUI::ObjectLayers::update_layers_list`) which corrupts the memory, leading to a crash later. But I don't see where exactly that would be. @YuSanka might.

combolek, 2021-02-27:
Not sure how much this helps but this is valgrind report for this crash. It kind of confirms your theory, though this is access after free (double free may come later?). But I don't know much about GUIs to debug any further.
```
==1007370== Invalid read of size 1
==1007370==    at 0x5D176A5: wxTextCtrl::GTKProcessEvent(wxEvent&) const (textctrl.cpp:2212)
==1007370==    by 0x5CA01DB: wxWindow::GTKHandleFocusOutNoDeferring() (window.cpp:4549)
==1007370==    by 0x5CA0797: wxWindow::GTKHandleFocusOut() (window.cpp:4504)
==1007370==    by 0x5C997EF: gtk_window_focus_out_callback (window.cpp:2057)
==1007370==    by 0x66943DA: _gtk_marshal_BOOLEAN__BOXED (gtkmarshalers.c:83)
==1007370==    by 0x6FE8E29: g_closure_invoke (gclosure.c:810)
==1007370==    by 0x70124F1: signal_emit_unlocked_R.isra.0 (gsignal.c:3809)
==1007370==    by 0x7005A6D: g_signal_emit_valist (gsignal.c:3505)
==1007370==    by 0x7006122: g_signal_emit (gsignal.c:3551)
==1007370==    by 0x665A073: gtk_widget_event_internal.part.0.lto_priv.0 (gtkwidget.c:7808)
==1007370==    by 0x6655D98: gtk_widget_send_focus_change (gtkwidget.c:16240)
==1007370==    by 0x6668A6F: do_focus_change (gtkwindow.c:8451)
==1007370==  Address 0x1985ec70 is 336 bytes inside a block of size 1,568 free'd
==1007370==    at 0x483B0B4: operator delete(void*, unsigned long) (vg_replace_malloc.c:593)
==1007370==    by 0x10C883C: Slic3r::GUI::LayerRangeEditor::~LayerRangeEditor() (GUI_ObjectLayers.hpp:49)
==1007370==    by 0x5E449A7: wxWindowBase::Destroy() (wincmn.cpp:578)
==1007370==    by 0x5E16899: wxSizerItem::DeleteWindows() (sizer.cpp:593)
==1007370==    by 0x10C481F: Slic3r::GUI::ObjectLayers::update_layers_list() (GUI_ObjectLayers.cpp:208)
==1007370==    by 0x10C495C: Slic3r::GUI::ObjectLayers::UpdateAndShow(bool) (GUI_ObjectLayers.cpp:231)
==1007370==    by 0x107BE3C: Slic3r::GUI::ObjectList::part_selection_changed() (GUI_ObjectList.cpp:3032)
==1007370==    by 0x10818AC: Slic3r::GUI::ObjectList::select_item(wxDataViewItem const&) (GUI_ObjectList.cpp:3937)
==1007370==    by 0x107FB7E: Slic3r::GUI::ObjectList::edit_layer_range(std::pair<double, double> const&, std::pair<double, double> const&, bool) (GUI_ObjectList.cpp:3634)
==1007370==    by 0x10C3796: Slic3r::GUI::ObjectLayers::create_layer(std::pair<double, double> const&, Slic3r::GUI::ObjectLayers::PlusMinusButton*, Slic3r::GUI::ObjectLayers::PlusMinusButton*)::{lambda(double, bool, bool)#3}::operator()(double, bool, bool) const (GUI_ObjectLayers.cpp:102)
==1007370==    by 0x10C7459: bool std::__invoke_impl<bool, Slic3r::GUI::ObjectLayers::create_layer(std::pair<double, double> const&, Slic3r::GUI::ObjectLayers::PlusMinusButton*, Slic3r::GUI::ObjectLayers::PlusMinusButton*)::{lambda(double, bool, bool)#3}&, double, bool, bool>(std::__invoke_other, Slic3r::GUI::ObjectLayers::create_layer(std::pair<double, double> const&, Slic3r::GUI::ObjectLayers::PlusMinusButton*, Slic3r::GUI::ObjectLayers::PlusMinusButton*)::{lambda(double, bool, bool)#3}&, double&&, bool&&, bool&&) (invoke.h:60)
==1007370==    by 0x10C6B88: std::enable_if<is_invocable_r_v<bool, Slic3r::GUI::ObjectLayers::create_layer(std::pair<double, double> const&, Slic3r::GUI::ObjectLayers::PlusMinusButton*, Slic3r::GUI::ObjectLayers::PlusMinusButton*)::{lambda(double, bool, bool)#3}&, double, bool, bool>, std::enable_if>::type std::__invoke_r<bool, Slic3r::GUI::ObjectLayers::create_layer(std::pair<double, double> const&, Slic3r::GUI::ObjectLayers::PlusMinusButton*, Slic3r::GUI::ObjectLayers::PlusMinusButton*)::{lambda(double, bool, bool)#3}&, double, bool, bool>(bool&&, (Slic3r::GUI::ObjectLayers::create_layer(std::pair<double, double> const&, Slic3r::GUI::ObjectLayers::PlusMinusButton*, Slic3r::GUI::ObjectLayers::PlusMinusButton*)::{lambda(double, bool, bool)#3}&)...) (invoke.h:113)
==1007370==  Block was alloc'd at
==1007370==    at 0x4839E7D: operator new(unsigned long) (vg_replace_malloc.c:342)
==1007370==    by 0x10C3AD8: Slic3r::GUI::ObjectLayers::create_layer(std::pair<double, double> const&, Slic3r::GUI::ObjectLayers::PlusMinusButton*, Slic3r::GUI::ObjectLayers::PlusMinusButton*) (GUI_ObjectLayers.cpp:103)
==1007370==    by 0x10C48B7: Slic3r::GUI::ObjectLayers::update_layers_list() (GUI_ObjectLayers.cpp:217)
==1007370==    by 0x10C495C: Slic3r::GUI::ObjectLayers::UpdateAndShow(bool) (GUI_ObjectLayers.cpp:231)
==1007370==    by 0x107BE3C: Slic3r::GUI::ObjectList::part_selection_changed() (GUI_ObjectList.cpp:3032)
==1007370==    by 0x106C503: Slic3r::GUI::ObjectList::selection_changed() (GUI_ObjectList.cpp:736)
==1007370==    by 0x1065F77: Slic3r::GUI::ObjectList::ObjectList(wxWindow*)::{lambda(wxDataViewEvent&)#1}::operator()(wxDataViewEvent&) const (GUI_ObjectList.cpp:181)
==1007370==    by 0x1094DD6: wxEventFunctorFunctor<wxEventTypeTag<wxDataViewEvent>, Slic3r::GUI::ObjectList::ObjectList(wxWindow*)::{lambda(wxDataViewEvent&)#1}>::operator()(wxEvtHandler*, wxEvent&) (event.h:536)
==1007370==    by 0x589751D: UnknownInlinedFun (event.cpp:1425)
==1007370==    by 0x589751D: wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) (event.cpp:1400)
==1007370==    by 0x5898CBD: wxEvtHandler::SearchDynamicEventTable(wxEvent&) (event.cpp:1897)
==1007370==    by 0x589906F: wxEvtHandler::TryHereOnly(wxEvent&) (event.cpp:1618)
==1007370==    by 0x589911A: UnknownInlinedFun (event.h:3927)
==1007370==    by 0x589911A: UnknownInlinedFun (event.h:3925)
==1007370==    by 0x589911A: wxEvtHandler::ProcessEventLocally(wxEvent&) (event.cpp:1555)
```

bubnikv, 2021-06-03:
@YuSanka could reproduce your issue, but she was not able to fix it after couple of days of effort. A volunteer help would be welcome.

closed 2026-09-29 by prusaslicer-bot[bot]