prusa3d/PrusaSlicer #6135: make mmu objects over air do bridges instead of floating perimeters

Photogad, 2021-02-26:
### Version
2.3.0

### Operating system type + version
win64

### 3D printer brand / version + firmware version (if known)
mk3s mmu2s

### Behavior
When there is a mmu object inside of another object that has space under it, the prusaslicer will make the new color for that mmu object start as perimeters in the air instead of doing a bridging across the gap, this makes the print impossible

_Is this a new feature request?_
Maybe

#### Project File (.3MF) where problem occurs
see below

[image]
[image]

[Cover MMU 3.zip](https://github.com/prusa3d/PrusaSlicer/files/6046553/Cover.MMU.3.zip)

bubnikv, 2021-02-26:
So you want PrusaSlicer to not extrude the outer perimeters for the yellow
object in that case, right?

Photogad, 2021-02-26:
> So you want PrusaSlicer to not extrude the outer perimeters for the yellow object in that case, right? 

A bridges of that color with no perimeters is fine, because this will all be internal, infill above it. 

Let's imagine the yellow color was the gray, it would not make perimeters, it would make a bridges with anchors. I'm not sure if this is possible but it sure would make features like this come out better if it's possible.

Just posting an idea to explore. I'm sure the answer is the PrusaSlicer sees that yellow color as separate object, because it IS a 2nd object (STL) and wants to create perimeters all around.

But perhaps there's some way to trick the algorithm to put bridges here, with anchors, and bridge flow,... Not print in air. 😆

bubnikv, 2021-02-26:
It certainly makes sense and it is doable, it "just" adds some additional complexity to an already complex piece of software.

rtyr, 2021-02-26:
You can easily avoid this problem in this case if you design your parts like this:

[image]

Result:
[image]

Photogad, 2021-02-26:
> You can easily avoid this problem in this case if you design your parts like this:
> 
> [image]
> 
> Result:
> [image]

Thank you for that tip, I'm new to CAD myself so I will make that note for the future when designing. 👍

I still think if possible the mmu algorithm here should be explored some day. PrusaSlicer now used by thousands of people. I'm sure I won't be the last person with this confusion on this issue. Just making suggestions for improvement.  Thank you team for your work.

closed 2026-09-29 by prusaslicer-bot[bot]