prusa3d/PrusaSlicer #6136: Feature Request: "Cooling Clearance"

jonnieZG, 2021-02-26:
### Version
2.3.0

### Operating system type + version
Any

### 3D printer brand / version + firmware version (if known)
All FDM printer models, all versions including 2.3.0

### Behavior
* When printing small parts, the heat from the head does not allow a printed layer to cool down, no matter how much you slow down the printing speed.
* Using a fan is not acceptable for some materials like ASA and ABS, since it also cools down the newly placed filament, which hinders layer bonding.
* This can usually be mitigated by printing two parts that are placed far enough from each other so a printed layer can have enough time away from the print head to cool down, but it unnecessarily slows down printing and causes excessive filament waste (cost and ecological issue)

### Proposed Behavior
* In the "Cooling" section add a parameter "Cooling clearance (mm)" that defines what is the minimal distance between a printed material and a print head, that counts as "cooling time" for that section. The value should be defaulted to ~40-50mm. 
* For example, if the head prints a layer, but there is a spot on that layer that has never been away from the head more that defined by the "Cooling clearance", the head should move away from the print in any possible direction (since this occurs in small prints, there should always be enough space for that) and wait for example 15 seconds (as defined by the value "Slow down if layer print time is below..."), then continue printing the next layer.

#### Project File (.3MF) where problem occurs
For example, create a simple 3mm wide and 2cm high cylinder and try printing it in ABS or ASA.

rtyr, 2021-02-26:
It seems a little bit overcomplicated to me.

You can usually use slow fan speed, it can be activated automatically below specified layer time threshold and disabled for rest of the print.

jonnieZG, 2021-02-26:
> It seems a little bit overcomplicated to me.
> 
> You can usually use slow fan speed even for ABS/ASA, it can be activated automatically below specified layer time threshold and disabled for rest of the print.

As I said, fan heavily affects layer bonding especially in ASA, as it primarily cools down the melted filament of the current layer. It should however only let the PREVIOUS layer to cool down. The fan for ASA should be used at bare minimum speed, just to prevent fan shroud from melting.

The only effective way to mitigate this issue is, as I said, printing two parts placed far away from each other. This feature would be a better solution.

rtyr, 2021-02-26:
We are using constant 20% fan speed for our ASA without issues.

jonnieZG, 2021-02-26:
> We are using constant 20% fan speed for our ASA without issues.

Can you print a single 3mm wide cylinder with these settings to come out properly?

rtyr, 2021-02-26:
> the head should move away from the print in any possible direction (since this occurs in small prints, there should always be enough space for that) and wait for example 15 seconds

There is similar feature in Cura, called "lift head", also requested here https://github.com/prusa3d/PrusaSlicer/issues/1342. 

But I don't think it is a good idea to wait somewhere for 15s. There will be certainly some heavy oozing and that oozed material will be missing after you return to the part...

jonnieZG, 2021-02-26:
> There is similar feature in Cura, called "lift head", also requested here #1342.
>
> But I don't think it is a good idea to wait somewhere for 15s. There will be certainly some heavy oozing and that [...]

Yes, oozing will occur if the pause is too long. Then add another setting such as "Head away time" that could be set to an arbitrary value.

jonnieZG, 2021-02-26:
It is a better choice to move the head in the XY plane rather than at Z axis. It is much faster.

rtyr, 2021-02-26:
I think that the "lift head" or similar feature can be usefull in some specific cases, but there should be probably some sort of optional priming (probably some tower) after such pause. I don't think that "cooling clearance" is worth of added complexity.

jonnieZG, 2021-02-26:
> I think that the "lift head" or similar feature can be usefull in some specific cases, but there should be probably some sort of optional priming (probably some tower) after such pause. I don't think that "cooling clearance" is worth of added complexity.

You are right. Perhaps instead of just waiting and allowing the filament to start oozing, a thin wipe tower of the same height should be generated automatically.

ledvinap, 2021-02-26:
Using wipe tower + wait outside print may be best way to handle short layer time in some cases.
Most of the logic is already implemented, all that is necessary is to move somewhere to wait (possibly with custom G-code with necessary delay as parameter) and then use wipe tower to prime extruder.

batata004, 2021-12-20:
@dartrax I think you provided the best solution in all the threads about this subject! Instead of creating a tower, instead of waiting and retracting filament... what about increasing the infill area on layers where the printing time is below a threshold? I think this is awesome cause at least the "extra" filament is being printed on some useful area (instead on a tower). And I think this solution of yours should be pretty easy to implement!

closed 2026-09-29 by prusaslicer-bot[bot]