prusa3d/PrusaSlicer #6137: compile  dependent  problem

wuhongjing929, 2021-02-26:
Hey friend. I ran into this problem when I compiled the dependent library ！
the  follow!
Building without Win10 Netfabb STL fixing service support
CMake Error at D:/software/cmake/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Boost (missing: filesystem thread log locale regex chrono
  atomic date_time) (found version "1.60.0")
Call Stack (most recent call first):
  D:/software/cmake/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  D:/software/cmake/share/cmake-3.20/Modules/FindBoost.cmake:2344 (find_package_handle_standard_args)
  CMakeLists.txt:269 (find_package)

robustini, 2021-03-27:
Same problem here:

-- Found Threads: TRUE
CMake Error at C:/Program Files/CMake/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:165 (message):
  Could NOT find Boost (missing: Boost_INCLUDE_DIR system filesystem thread
  log locale regex chrono atomic date_time iostreams) (Required is at least
  version "1.66.0")
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:458 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.18/Modules/FindBoost.cmake:2177 (find_package_handle_standard_args)
  CMakeLists.txt:322 (find_package)

robustini, 2021-03-31:
Boost it is not covered in the documentation explaining how to compile, but wants it installed on the system, and not just him.
Please update the docs.

bubnikv, 2021-10-05:
Docs updated. Closing

closed 2021-10-05 by bubnikv