prusa3d/PrusaSlicer #6138: position of pause print

Intox240, 2021-02-26:
### Version
Version 2.3.0+win64

### Operating system type + version
Windows 10 (build 19041), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S+
3.9.3-3556

### Behavior
The layer placement of the pause print is confusing in my opinion
Expected: I move the slider on the right to the point where I want the printer to pause, so that the top layer seen in the slicer is the same as on the printer at the time of the pause.
Reality: The pause is placed one layer lower than currently shown. this might cause problems with crashes if I place parts into the print that have to be very precise as they now protrude 1 layer. 
In addition, the layer above the current layer is displayed in grey. I would expect the grey layer to be the one printed directly after the pause, when in fact the orange one directly below the grey one will be printed after the pause.

layer that is displayed at the height where the pause is added
[image]

whole print, grey layer on top
[image]

print at the point of the pause
[image]

I found this issue reproduceable within the last 2 days, the test with the pictures was done today. Sadly I closed the project, without saving, and when I wanted to reproduce it to save the project to attach it, for some reason at least the grey layer seems to be displayed diffenrently. Since I currently have a long print running, I can't test where the pause will end up on the printer for the attatched project. The gcode is from the print in the picture.

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
[pausetest.zip](https://github.com/prusa3d/PrusaSlicer/files/6048527/pausetest.zip)

Intox240, 2021-02-26:
since it is not that easy to see: 1st picture: 3 layers, 2nd Picture all 4 layers, 3rd picture(printed): 2 layers

rtyr, 2021-02-26:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5780. Closing.

closed 2021-02-26 by rtyr