prusa3d/PrusaSlicer #6143: PrusaMini Move after Print

Swissi78, 2021-02-26:
### Version
2.3.0

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa Mini+ Firmware 4.3.1

### Behavior
If the printing is ended with the End GCode, it does not stop, but the Z axis goes down to 1, at the same time the x axis goes to 40, only then does the Mini stop printing. According to GCode, X should stay at 178 and Z at 60, it goes there first. I used the original profile from the Prusaslicer for the Mini. I discussed the problem with Live Support and when I slice the same file with version 2.2.0, the GCode works as it should. 

_Is this a new feature request?_
The problem only appeared with the configuration update from 02/21/2021, the prints I made before that worked correctly.

[https://1drv.ms/v/s!AjefiSaTeX0V6D7BTKilo8Fbnbkv?e=AIC0Sj](url)

rtyr, 2021-02-26:
I don't understand your issue. Please provide the problematic g-code.

Swissi78, 2021-02-26:
The problem is very simple, the printer moves to the position of the end GCode after printing, but then does not stop, but z moves down to 1 and x moves to 40, although x should stop at 178 and Z over to 30 max z from prind object 

[Prusa_Base_0.07mm_PETG_MINI_6h53m.zip](https://github.com/prusa3d/PrusaSlicer/files/6052644/Prusa_Base_0.07mm_PETG_MINI_6h53m.zip)

it does not do it with this gCode, because after the end Gcode it remains at the position that is also specified in the code.

[Test_5mm_0.2mm_PETG_MINI_10m.zip](https://github.com/prusa3d/PrusaSlicer/files/6052658/Test_5mm_0.2mm_PETG_MINI_10m.zip)

Swissi78, 2021-02-26:
you can see the issue here:
https://1drv.ms/v/s!AjefiSaTeX0V6D7BTKilo8Fbnbkv?e=jGzZqB

rtyr, 2021-02-26:
This is firmware issue. Likely related to https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/733.

I already saw very similar behavior, reported by some different customer. It was caused by some additional commands/comments in the start g-code (related to Bondtech upgrade if I remember correctly). I can see, you are also using Bondtech upgrade and your start g-code is not stock.

Anyway, the provided g-code is OK (marked line is the last move). This is not PrusaSlicer issue, so I am closing it here.
[image]

closed 2021-02-26 by rtyr

Swissi78, 2021-02-26:
ok, the funny thing is if I create a Gcode with the same firmware but with Prusaslicer 2.2.0, I don't have the problem ...

rtyr, 2021-02-26:
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/733#issuecomment-748618288