prusa3d/PrusaSlicer #6144: CR-10 Max profile?

WimVDK, 2021-02-26:
### Version
2.3.0+win64

### Operating system type + version
Win 10 Pro x64

### 3D printer brand / version + firmware version (if known)
None since there's no profile for my printer

### Behavior
No behavior as there's nothing to configure. No printer profile exists for my printer

_Is this a new feature request?_
Yes, for a printer config profile for the one i have. The CR-10 Max

#### Project File (.3MF) where problem occurs
No projects exists as i can't print anything with prusaslicer

rtyr, 2021-02-26:
3rd party printer profiles are added based on the community contributions. That being said, the CR-10 Max profile is already prepared by @pmjdebruijn for next PrusaSlicer release. Closing.

closed 2021-02-26 by rtyr

WimVDK, 2021-02-27:
Any idea when that may be (not asking for a specific date, just a timeframe)? Or which profile i could try in the mean time

pmjdebruijn, 2021-02-27:
Just select the CR-10 and change the max print height to 470 and bedshape 440x440 at -5x-5 and you should be fine if I'm not mistaken.

WimVDK, 2021-02-27:
why only 440 and those -5? The CR-10 max has an advertised volume of 450*450 on the bed (Still new at this, only have this thing for 3 months now, and it's my first (and currently only) one)

pmjdebruijn, 2021-02-27:
There is prime line being made on the left, that must be avoided, on the other edges the distance is kept, so the center remains the center, and usually it helps avoid the nozzle bump into any clips holding a glass bed.

Creality is very inconsistent in real-world bed size vs advertised usable bed-size vs firmware motion addressable bed size.