prusa3d/PrusaSlicer #6147: First layer skips part of the STL model

guivo, 2021-02-27:
I am using PrusaSlicer 2.3.0, Windows 64-bit version downloaded from github.

I have tried to slice an STL model, describing not a single solid but few elements. The gcode produced by the PrusaSlicer skip several small parts during the first layer, causing the print to fail.

The model is here:
[image]

The screenshot of the first layer is as follow
[image]
6 elements are missing, but they are printed since the second layer:
[image]

The visualization is not an artifact and the printer also started printing the elements since the 2nd layer, with clear adhesion problem.

[bn-click-feet-100mm-12nut-5mm.zip.zip](https://github.com/prusa3d/PrusaSlicer/files/6055482/bn-click-feet-100mm-12nut-5mm.zip.zip)

The original STL can be found here: https://cdn.thingiverse.com/assets/6b/3b/03/45/ae/bn-click-feet-100mm-12nut-8mm.stl.

neophyl, 2021-02-27:
Theres definitely something strange going on.  When I look at the model in Blender the 12 hexagons are built up of three rows of 4.  They actually share geometry so they touch at the thin edges.  So there's basically really thin contact points between them in their row of 4.  That seems to be causing PS some issues when Elephants foot is used.  If you reduce the elephants foot compensation to zero your original amf slices with all the first layers.  The original stl does too.  

The weird thing is that when sliced without elephants foot, the first layers are all there but the hexagons do NOT touch each other, which the original they do.

guivo, 2021-02-28:
Thanks @neophyl, I am new to PS and I learnt an interesting option. To finalize the print, luckily the designer shares the SCAD file and I made a STL model without the hexagons and this is printing as expected (the small pieces are not really useful to me).

I was sharing the issue to check if this highlights some small issue with PS, as I said new user with a creality machine, I am trying PS and  it looks great, the issue seemed however quite serious.

closed 2026-09-29 by prusaslicer-bot[bot]