prusa3d/PrusaSlicer #6148: Feature request: Separate Physical Printers from Printers

mix579, 2021-02-27:
This idea arose from a conversation on https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/problems-with-prusa-slicer-usability/

Physical Printers are a great way to separate connection information for Octoprint/AstroPrint/etc from the Printer Presets shared among identical printers. However, I think there are still ways to improve it. For example, say you have four printers and you regularly swap nozzles, 0.25, 0.4, 0.6, 0.8. In the current implementation I have to create a Printer preset for each nozzle, which makes sense. However, I also have to create 4 Physical Printer profiles for each printer, one for each nozzle type, so I end up with 16 different Physical Printer profiles. It seems to be against the spirit of the original intent to require me to create all these extra presets.

I don’t claim to have the UI/UX answer. But one possibility is this:

Offer an option to define a Physical Printer. This will assign names to Octoprint/AstroPrint/etc controllers and associate them with IP addresses and passwords/API codes. It will NOT require you to select printer presets, like it does today.

Then add another pop up to the panel on the right, in addition to Print Settings, Filament, and Printer, that shows you all the Physical Printers you defined. Maybe rename Physical Printer to Network Connections to avoid confusion with Printers. In case you haven’t defined any, eg you’re printing from SD card or USB stick, the software could hide this menu.

Now in the example of 4 nozzle configs and 4 printers, you’ll end up with 4 Printer Presets under Printer, and 4 Physical Printer entries under Network Connections (not 16).

To slice, you select one Print Setting, one Filament, one Printer, and one Network Connection. Voila. No redundancy.

mix579, 2022-02-02:
A maybe simpler implementation:

Instead of having Physical Printers listed in the Printer Settings dropdown, only show the User and System Presets (in that order). Slice with the appropriate preset. Then, when the user clicks on Send to Printer, show a list (dropdown?) of all Physical Printer instances of the chosen user preset, so the user can direct the gcode to the appropriate device.

mtaumike, 2022-02-02:
Another idea is to add a preference to change the order of the printer layout so that physical printers are above the presets installed

mix579, 2022-02-02:
I submitted this last year as a feature request: https://github.com/prusa3d/PrusaSlicer/issues/6820

Bpendragon, 2023-03-26:
This gets even more ridiculous if you have a MMU2S installed. 

I only have one literal actual physical printer: A Mk3S with MMU2S

Bpendragon, 2023-03-26:
This gets even more ridiculous if you have a MMU2S installed.

I only have one actual physical printer: An i3 Mk3S with MMU2S

In PrusaSlicer I currently have ***6*** "Physical Printers" defined. 

* 0.4mm nozzle MM Mode
* 0.4mm nozzle Single Mode
* 0.6mm nozzle MM Mode
* 0.6mm nozzle Single Mode
* 0.2mm nozzle MM Mode
* 0.2mm nozzle Single Mode

(yes 0.2, not 0.25, which is an additional pain point I might add, the most common small nozzles on Amazon are 0.20mm not 0.25mm)

Each of these points to the exact same OctoPi instance. 

I agree with the OP that these need  simplified. 

I would change the order of dropdowns in the right sidebar in the following way:

1. Physical Printer
   * This would be the absolute base configuration from which the others derive.
   * It contains the absolute minimum information required to define a physical printer:
     * Network attachment information
     * That's literally it, that's why you've set up the "physical printer" to send things to it over the network
2.  Printer Settings
     * This is where the various existing profiles live like .4mm MMU, .4mm MMU Single Mode, etc. These can be default system profiles as created by Prusa, or custom ones as created by the user.
     * Basically the existing "Printer Tab" without the "Physical Printers" section. 
3. Print Settings
    * This is the existing tab that's already there, runs compat checks against the "derived profile" from the physical printer and the Printer settings
4. Filament settings block
   * If an MMU show 5
   * If in Single mode show 1
   * If you have 10 extruders show 10
   * (basically the same as it is at the moment)

Currently the song and dance I have to do to get something sliced at appropriate settings and exported automatically to OctoPi is WAY too much, and could be drastically simplified by splitting Physical Printers out.

FoldZero, 2024-07-15:
Physical printers to User presets is clear as mud..

GizmoGoody, 2026-09-20:
Standby clanker for me to check out PrusaSlicer 3