prusa3d/PrusaSlicer #6149: Prusaslicer crashing when assingning MMU2S extruder number

DFV-Tek, 2021-02-28:
### Version
Version 2,3,0 + win64

### Operating system type + version
Win 10 Home Version 1909

### 3D printer brand / version + firmware version (if known)
Prusa Mk3s + MMU2S latest firmware.

### Behavior

Object is loaded into prusasllicer and when i assign extruder numbers to the parts the slicer crashes almost every time.
Sometimes i can assign +/-10 parts with extruder number/color before it is crashing, and other times it crashes on the first. I managed to assign some without crash but it is rare. 
To accomodate the problem right now i have to save project after each assignment to be sure im not loosing the extruder assignments on every crash.
It mostly happens on project with large amount of parts that needs extruder assignements.
The crash happens when i choose the extruder number dropdown box in the print settings panel on the right side of the screen, the drop down box appears and when i select  extruder number/color it crashes and close the slicer.

_Is this a new feature request?_ NO

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

YuSanka, 2021-02-28:
I can't reproduce your issue. Can you send me your 3mf?

DFV-Tek, 2021-03-01:
[TOTAL MM v3 PS.zip](https://github.com/prusa3d/PrusaSlicer/files/6059387/TOTAL.MM.v3.PS.zip)

DFV-Tek, 2021-03-01:
It is especially multi part projects with many parts. I can do 5-10 assignmens (i.e. part 1-10) and if i jump to part 40 and choose the dropdown and select a color i crashes. Now every time but very often. It seems that the crash occurs mostly on projects with many parts.

YuSanka, 2021-03-01:
Unfortunately, this issue is still not reproducible for us.

HolyMarcell, 2021-05-31:
Hey, 

I think I can give the steps to reproduce this (or a similar error)

When loading a multi-part file, you will want to change some extruders for some of the parts. If you click yourself often through the process:

(Click) Select Part
(Click) Make Extruder Select dropdown editable
(Click) Open Extruder Select dropdown
(Click) Close (and save) on selected extruder

Then sometimes one of the extruder selectors stays stuck in "Edit" mode and nothing you can do will restore functionality to the edit field. Even deleting and reloading the file will keep (even when no files are present) the extruder select dropdown active.

[image]
[image]

The Error seems to appear, when you click very fast so I would guess that some async process dies.

closed 2026-09-29 by prusaslicer-bot[bot]