prusa3d/PrusaSlicer #6151: Support enforcer is not generating

aj1004, 2021-03-01:
### Version 
_Version of PrusaSlicer used goes here_
2.3.0+

### Operating system type + version
Mac OS Big Sur 11.2.2

### 3D printer brand / version + firmware version (if known)
Prusa MK3S+

### Behavior
Support enhancer is not working properly when I load files

#### Project File (.3MF) where problem occurs
[Archive.zip](https://github.com/prusa3d/PrusaSlicer/files/6059351/Archive.zip)

aj1004, 2021-03-01:
Any suggestions when support enhancer is not generated?
Please let me know. Thank you bunch.

neophyl, 2021-03-01:
Your zip just contains stl files.  A saved project .3mf file is needed so that people can see exactly how you are placing your enforcers and what settings you are using.

aj1004, 2021-03-01:
Here is 3mf file. if you look there is no calculation path for enforcer file. Any idea?
[Enforcer calculation missing.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6060009/Enforcer.calculation.missing.3mf.zip)

neophyl, 2021-03-01:
OK for a start your project settings have supports turned OFF.  So it doesn't matter what enforcers you place.  No supports.  You need to change the setting to Supports: For Support Enforcers Only if you want to control their placement using enforcers.

Secondly your enforcer is under an area that, if the walls were to print successfully, could be bridged so wouldnt need support.  To make it add supports to that area you would need to turn OFF  -Print Settings>Support Material> Dont Support Bridges (untick it).

Thirdly  if you enable support for enforces AND turn off Dont support bridges you STILL dont get support for that area as your enforcer modifier doesnt actually touch the model above it.  When  you move the enforcer up to actually touch the model you then get supports.  An enforcer (or blocker) only work where they intersect with the model surface.

Also you have 4 small contact points with everything else needing support, none of which have any support enforcer placed under them.  You need to place enforcers under all the perimeters, that large section sticking out the rear and the entire middle section as none of those touch the build plate.  Without that you are also printing in midair.

aj1004, 2021-03-01:
Worked perfect. Thank you so much.

closed 2021-03-09 by rtyr