prusa3d/PrusaSlicer #6153: Discontinuous perimeter with retracts

haarp, 2021-03-01:
Hello!

### Version
2.3.0

### Operating system type + version
Gentoo Linux

### 3D printer brand / version + firmware version (if known)
MK3 3.8.2

### Behavior
In curved perimeters, certain thicknesses seem to make it go discontinuous, making the slicer add many short segments with individual retracts, sometimes even pritning them "out of order".

[image]

#### Project File (.3MF) where problem occurs
[PartT.zip](https://github.com/prusa3d/PrusaSlicer/files/6061266/PartT.zip) (rename from .zip to .3mf first)

Check layer 39.12.

Thanks!

rtyr, 2021-03-01:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/3025, etc.

Enabling "detect thin walls" may help in this case.

haarp, 2021-03-01:
> Similar to #3025

I'm not 100% sure. I've seen #3025 many times before, but never with retracts.

> Enabling "detect thin walls" may help in this case.

Interestingly, it does! The discontinuous internal perimeter is then replaced with gap fill. But the documentation for "detect thin walls" seems incomplete then, as one wouldn't expect it to have an effect on this 3-perimeter structure. Hmm.

closed 2026-09-29 by prusaslicer-bot[bot]