prusa3d/PrusaSlicer #6158: Feature request: Show remote upload queue automatically when sending to Octoprint

shogun168, 2021-03-01:
### Version
NA

### Operating system type + version
NA

### 3D printer brand / version + firmware version (if known)
NA

### Behavior
NA

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
Usability improvement: Automatically show the remote upload queue when sending something to Octoprint. Right now, there's no notification or signal that the upload has completed and on a few occassions, I have closed the application without the upload finishing. Even just showing a mini progress bar on the toolbar would help rather than having to bring up the remote queue every time.

tknx, 2021-03-29:
I would like 2-way communication fully - it would be great to send prints and have them act similar to as if the printer was directly connected.

bubnikv, 2021-03-29:
We will show notifications with progress bar and cancel button in
PrusaSlicer 2.4. The code is already in master.

shogun168, 2021-03-29:
Awesome, can't wait to check it out.

bubnikv, 2021-09-25:
Uploads are shown by notifications in PrusaSlicer 2.4.0-alpha1.
Closing.

closed 2021-09-25 by bubnikv