prusa3d/PrusaSlicer #6159: Loss of quality/detail on spliced model

reedy, 2021-03-01:
### Version
2.3.0+

### Operating system type + version
```
PrusaSlicer
Version:   2.3.0+
Build:     PrusaSlicer-2.3.0+-202101111304

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Catalina Version 10.15.7 (Build 19H524)
Total RAM size [MB]: 17,180MB
OpenGL installation
GL version:   2.1 INTEL-14.7.12
Vendor:       Intel Inc.
Renderer:     Intel(R) Iris(TM) Graphics 6100
GLSL version: 1.20
Installed extensions:

GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
```

### 3D printer brand / version + firmware version (if known)
Prusa Mini 4.3.1

### Behavior
Create an STL from https://skyline.github.com/, for example mine from 2008 is https://skyline.github.com/reedy/2008

[reedy-2008.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/6065063/reedy-2008.stl.zip)

Import this STL into PrusaSlicer, all looks good on the initial rendering (same as it does on the skyline site).

[image]

Slice it, and look at the rendering in preview:

[image]
[image]

The detail on some of the lettering has semingly gone.

After importing, it is noted that there are thousands of auto repaired errors
[image]

I don't really know enough about 3D printing to know if this is a problem with the underlying model or what PrusaSlicer is doing with it. If you "upload" it to Shapeways via the Github site, their rendering (though from what stage, I don't quite know) looks ok. I've no idea what happens if I actually requested it printing.

However, if you look at https://twitter.com/MishManners/status/1362181990473109505 and more specifically https://twitter.com/MishManners/status/1362184369612365826 the prints others have had done seem alright in the end...

#### Project File (.3MF) where problem occurs
[reedy-2008.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6065064/reedy-2008.3mf.zip)

rtyr, 2021-03-02:
The model has a lot of errors, PrusaSlicer is just not able to automatically repair all of them. It slices fine with repaired model (attached).

[image]

[reedy-2008_repaired.zip](https://github.com/prusa3d/PrusaSlicer/files/6067188/reedy-2008_repaired.zip)

reedy, 2021-03-02:
Thanks. Out of interest, how did you repair it?

I guess I'll try and contact github support and see if they can do anything to fix the models they're distributing in the first place...

rtyr, 2021-03-02:
Netfabb.

reedy, 2021-03-07:
FWIW, I did try and report this to GitHub support... No idea if it'll yield anything though

>Thank you for sharing your thoughts, comments, and ideas with us! I have forwarded your ticket to the Product team who reads and evaluates all feedback, however we cannot guarantee a response to every submission.
>
>Please note that this means the Developer Support team will no longer be able to see changes to the ticket, so if you have a follow up that requires further support, please create a new ticket. Thank you again for your feedback!

bubnikv, 2021-09-22:
This model has really many open (disconnected) edges.

[image]

jepler, 2022-01-03:
The free ["admesh" commandline program](https://admesh.readthedocs.io/en/latest/) seems to have given me a printable file. Typical commandline & output:
```
$ admesh jepler-2021.stl  -b 2021-fixed.stl
ADMesh version 0.98.4, Copyright (C) 1995, 1996 Anthony D. Martin
ADMesh comes with NO WARRANTY.  This is free software, and you are welcome to
redistribute it under certain conditions.  See the file COPYING for details.
Opening jepler-2021.stl
Checking exact...
Checking nearby. Tolerance= 0.000042 Iteration=1 of 2...  Fixed 0 edges.
Checking nearby. Tolerance= 0.015734 Iteration=2 of 2...  Fixed 8 edges.
Removing unconnected facets...
Filling holes...
Checking normal directions...
Checking normal values...
Calculating volume...
Verifying neighbors...
Writing binary file 2021-fixed.stl

================= Results produced by ADMesh version 0.98.4 ================
Input file         : jepler-2021.stl
File type          : ASCII STL file
Header             : solid stlmesh
============== Size ==============
Min X = -74.908424, Max X =  75.179634
Min Y = -16.008951, Max Y =  14.471186
Min Z = -4.486088, Max Z =  29.699999
========= Facet Status ========== Original ============ Final ====
Number of facets                 : 10427               10288
Facets with 1 disconnected edge  :   411                   0
Facets with 2 disconnected edges :   206                   0
Facets with 3 disconnected edges :    24                   0
Total disconnected facets        :   641                   0
=== Processing Statistics ===     ===== Other Statistics =====
Number of parts       :   385        Volume   :  49554.738281
Degenerate facets     :   528
Edges fixed           :     8
Facets removed        :   556
Facets added          :   417
Facets reversed       :   556
Backwards edges       :     0
Normals fixed         :   608
```

reedy, 2022-01-03:
> They haven't fixed it yet.

Quelle surprise, no? :)

> The free ["admesh" commandline program](https://admesh.readthedocs.io/en/latest/) seems to have given me a printable file.

Thanks for the hint!

reedy, 2022-01-03:
It's curious how much smaller the file size is after running through admesh

```
-rw-r--r--@   1 reedy  staff   709784  3 Jan 19:55 reedy-2008-fixed.stl
-rw-r--r--@   1 reedy  staff  4318214  3 Jan 19:47 reedy-2008.stl
-rw-r--r--@   1 reedy  staff   614184  3 Jan 19:57 reedy-2009-fixed.stl
-rw-r--r--@   1 reedy  staff  3756775  3 Jan 19:48 reedy-2009.stl
-rw-r--r--    1 reedy  staff   511184  3 Jan 19:57 reedy-2010-fixed.stl
-rw-r--r--@   1 reedy  staff  3102072  3 Jan 19:49 reedy-2010.stl
-rw-r--r--    1 reedy  staff   493084  3 Jan 19:57 reedy-2011-fixed.stl
-rw-r--r--@   1 reedy  staff  2916322  3 Jan 19:49 reedy-2011.stl
-rw-r--r--    1 reedy  staff   504984  3 Jan 19:57 reedy-2012-fixed.stl
-rw-r--r--@   1 reedy  staff  3004018  3 Jan 19:49 reedy-2012.stl
-rw-r--r--    1 reedy  staff   502884  3 Jan 19:57 reedy-2013-fixed.stl
-rw-r--r--@   1 reedy  staff  3061470  3 Jan 19:50 reedy-2013.stl
-rw-r--r--    1 reedy  staff   465284  3 Jan 19:57 reedy-2014-fixed.stl
-rw-r--r--@   1 reedy  staff  2914665  3 Jan 19:50 reedy-2014.stl
-rw-r--r--    1 reedy  staff   456984  3 Jan 19:57 reedy-2015-fixed.stl
-rw-r--r--@   1 reedy  staff  2997693  3 Jan 19:51 reedy-2015.stl
-rw-r--r--    1 reedy  staff   556384  3 Jan 19:57 reedy-2016-fixed.stl
-rw-r--r--@   1 reedy  staff  3569706  3 Jan 19:51 reedy-2016.stl
-rw-r--r--    1 reedy  staff   455084  3 Jan 19:57 reedy-2017-fixed.stl
-rw-r--r--@   1 reedy  staff  2920518  3 Jan 19:51 reedy-2017.stl
-rw-r--r--    1 reedy  staff   657584  3 Jan 19:57 reedy-2018-fixed.stl
-rw-r--r--@   1 reedy  staff  4130269  3 Jan 19:52 reedy-2018.stl
-rw-r--r--    1 reedy  staff   567584  3 Jan 19:57 reedy-2019-fixed.stl
-rw-r--r--@   1 reedy  staff  3567750  3 Jan 19:52 reedy-2019.stl
-rw-r--r--    1 reedy  staff   535184  3 Jan 19:57 reedy-2020-fixed.stl
-rw-r--r--@   1 reedy  staff  3189800  3 Jan 19:53 reedy-2020.stl
-rw-r--r--    1 reedy  staff   477584  3 Jan 19:57 reedy-2021-fixed.stl
-rw-r--r--@   1 reedy  staff  3001176  3 Jan 19:53 reedy-2021.stl
```

reedy, 2022-01-03:
Curiously on the 3D model view, the "fixed" version has something odd in the 0 in 2021.

[image]

On the spliced version it moves to the original.

[image]

Might be seperate task worthy... But is there something that can be integrated from https://github.com/admesh/admesh into the PrusaSlicer to be able to do more/better fixing itself?

bubnikv, 2022-01-04:
Admesh is integrated into PrusaSlicer, we just don't use the whole fixing functionality of admesh during STL import, because it was  doing more harm then good in most cases.

We may consider adding a "fix by admesh" menu item, or possibly offer each of the admesh fixing services as an individual menu item as in IdeaMaker.

[image]

However NetFabb does a much better job in most cases.

reedy, 2022-01-04:
> However NetFabb does a much better job in most cases.

With how much it costs, you'd kinda hope so ;)

reedy, 2022-03-06:
> We may consider adding a "fix by admesh" menu item, or possibly offer each of the admesh fixing services as an individual menu item as in IdeaMaker.

This does seem a good idea. Got some other models to print for a friend and had to lookup the repair commands again. Being able to do it from a menu in PrusaSlicer would make the workflow smoother

reedy, 2026-09-02:
https://skyline.github.com/ is now deprecated in favour of https://github.com/github/gh-skyline

[reedy-2008-github-skyline.stl.zip](https://github.com/user-attachments/files/31751216/reedy-2008-github-skyline.stl.zip)

[image]

I don't know how much is the change in the generated file, but I guess that's a good part of it.

If we re-import the original file I uploaded originally:

[image]