prusa3d/PrusaSlicer #6161: Feature request: export in silent mode

eiffel31, 2021-03-02:
### Version
PrusaSlicer 2.3.0

### Behavior
In printer settings, you can configure machine limits for 2 modes: normal mode and silent mode.
So, you can set speed, acceleration, jerk... for both mode

The generated g-code starts with setting printers limits for normal mode.

It would be great to have a toggle button in order to select silent mode, so the generated g-code exports the limits as defined for the silent mode.

An alternative could be to add "Silent g-code export" next to current "g-code export". 

Thanks!

Note: silent_mode setting is not for this purpose.

rtyr, 2021-03-02:
Silent mode machine limits were added primarily for our MK3 printer, which supports two modes (it can be switched directly on the printer). Machine limits for silent mode are there only for time estimate. It doesn't make sense to export those limits, because silent mode (stealthchop) can be activated on the printer at any time and machine limits are applied automatically by the firmware.

I guess you want use it for different purpose than it was intended for. Isn't it easier to just create separate "silent" printer profile?

eiffel31, 2021-03-03:
For all other printers that do not have a silent-mode setting on their screen, it would be very interesting to select the machine limits to generate at export time.
Choosing the normal profile will result in noisy but efficient printing.
Choosing the silent profile will result in quiet printing.

All the configuration values are already there, just a selector is needed to select the set to use.
For your printers with silent mode in firmware, this selector should select the default value to start with at printing time.

rtyr, 2021-03-03:
> Choosing the normal profile will result in noisy but efficient printing.
> Choosing the silent profile will result in quiet printing.

This can be easily done by creating separate printer profile.

eiffel31, 2021-03-03:
machine_limits_usage can be set to write limits in g-code, to use limits for computed estimations, to ignore limits.

This would be really natural to have 2 choices in place of "write limits in g-code":
- write normal limits in g-code
- write silent limits in g-code

Then PS will write set 0 or set 1 from the limits.

Thanks

rtyr, 2021-03-03:
What is better and faster?

Your suggestion
---------------
1) go to the printer settings
2) go to the machine limits
3) switch emit g-code value
4) slice

+ keep in mind what was your last setting

Current state
-------------
1) select printer profile (Normal or "Silent" variant)
2) slice

Anyway. As I mentioned, stealth mode values are meant for printers which supports stealth mode (as stated in the tooltip). 

[image]

closed 2021-04-12 by rtyr