prusa3d/PrusaSlicer #6162: Bridge Infill

Rhoenbussard, 2021-03-02:
### Version
PrusaSlicer-2.3.0+win64-20210111131

### Behavior
I noticed a special feature of the PrusaSlicer. Under the top layers there is a bridge infill with a height of 0.4mm corresponding to the nozzle diameter. But the layer height is only 0.1mm.

LAYER_CHANGE
;Z:9.1
HEIGHT:0.1
;Flow 1st layer
;Layer:(layer_num)

...

;TYPE:Bridge infill
;WIDTH:0.400395
;HEIGHT:0.4

At this point the wall is no longer printed closed. The change in the wall becomes clear from an EW of 0.33 and more and more gaps are in the wall.
I can't find a way to reduce the Bridge Infill height. I think the height cause too much plastic for a good print  with a low EW.
I
[image]
[image]

rtyr, 2021-03-02:
> Under the top layers there is a bridge infill with a height of 0.4mm corresponding to the nozzle diameter. 

That is correct.  See https://manual.slic3r.org/advanced/flow-math.

> I can't find a way to reduce the Bridge Infill height. 

You are extruding to the air, so round extrusion of the same diameter as nozzle diameter is expected.

rtyr, 2021-03-08:
Answered, no response. Closing.

closed 2021-03-08 by rtyr