prusa3d/PrusaSlicer #6167: Clarification request for bridge_flow_ratio

eiffel31, 2021-03-03:
### Version
PrusaSlicer 2.3.0

### Behavior
bridge_flow_ratio definition is missing some detail: does this setting override extrusion_multiplier when printing bridges or does this setting combines (multiply) with extrusion_multiplier when printing bridges?

I guess the tooltip should be more clear on that point.

Thanks

rtyr, 2021-03-03:
It is separate ratio for bridge flow. What is bridge flow: https://manual.slic3r.org/advanced/flow-math, chapter "Bridges: the easy case".

closed 2021-03-03 by rtyr

eiffel31, 2021-03-03:
I carefully read the link and your response but was not sure of PS behavior.

So I made some test and give here the result for other future users that may ask:
**the amount of filament to extrude for bridge is computed by multiplying bridge_flow_ratio and extrusion_multiplier**

So multiplying bridge_flow_ratio and extrusion_multiplier effects are combined together for bridges. bridge_flow_ratio is not a new ratio to apply in place of the standard one, it is an additional ratio being applied.