prusa3d/PrusaSlicer #6168: Integrated printer test objects?

peter12311, 2021-03-03:
### Version
_Version of PrusaSlicer used goes here_
2.3.0

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
Currently 5 printers, all highly customized, all on Marlin 1.1 or 2.x

### Behavior
No problem, but a question / request.
Probably anybody using 3d-printing prints multiple test objects as heattowers, leveling patterns, cubes, benchys etc.. Wouldn't it be nice to have a slicer option to genenate a couple of these objects directly in the slicer? i.e. Heattowers with a generated variable temperature, leveling patterns generated to fit the heatbed-size and so on. At least for me it's always loads of manual work including all possibility to miss one or the other setting accidentially.

Maybe something like this already exists - so please give me a hint. If not I'd love to see this as an additional feature in this great slicer.

_Is this a new feature request?_
yes

bttguy, 2021-03-03:
It would be a nice addition!

plampix, 2021-03-04:
I use SuperSlicer (a fork of PrusaSlicer) to do these kind of things, and port the changes back to my PrusaSlicer profiles.

Photogad, 2021-03-15:
As other user had said, it is already in SuperSlicer.  But I would like to see it in PrusaSlicer too, because overall this program is more stable on my system than SuperSlicer. Do a pull request? Prusa could make the test objects even better...

redPanther, 2021-11-09:
I would love to see this too. This feature can be linked with the great prusa wiki. In the wiki it is not needed to download test objects, just say .... choose this entry from calibration menu. Would love to see more complex tests like linear advance. Its a pain in the ass to mess around with existing generators/files/gcode when you have a modified printer. 

In addition I would like to see that you can add your own entries for calibration objects (e.g. like a benchy. Perhaps including it in the release could be problematic due to licensing issue?)

foreachthing, 2021-11-09:
> In addition I would like to see that you can add your own entries for calibration objects (e.g. like a benchy). 

You could add Benchy yourself: 
Right click on the build plate -> Add shape -> Gallery -> Add (then browse to your [downloaded Benchy file](https://www.3dbenchy.com/download/)).

redPanther, 2021-11-09:
> You could add Benchy yourself:
> Right click on the build plate -> Add shape -> Gallery -> Add (then browse to your [downloaded Benchy file](https://www.3dbenchy.com/download/)).

thats neat, thx!

CPHfx, 2021-11-09:
> > In addition I would like to see that you can add your own entries for calibration objects (e.g. like a benchy).
> 
> You could add Benchy yourself: Right click on the build plate -> Add shape -> Gallery -> Add (then browse to your [downloaded Benchy file](https://www.3dbenchy.com/download/)).

Not sure why that helps?
I think calibration right now is difficult, as you have to handle with own gcode, to set up a calibration tower.
For some calibrations like extrusion multiplier, k, or retraction, it is (for me) extremely difficult to set this up... heat tower is ok ;)

For adding a downloaded file to your bed, right now you just click on add file, browse to you folder, and add it.
If you want to open this dialog via right click, i think that would be another request and has not so much to do with calibration capabilities of PS

foreachthing, 2021-11-09:
@CPHfx  it helps because he/she can add custom models very quickly.
This has _nothing_ to do with calibrating, I agree! I myself use SuperSlicer for calibrating....

ibash, 2021-12-31:
Given the license of prusaslicer/superslicer, there's nothing stopping literally porting superslicer's calibration tests over.

ibash, 2022-01-01:
In progress, but I get a segfault...

[image]
[image]
[image]

poudenes, 2024-05-24:
Is this still is progress? I don't see the option in Prusa 2.7.4 MacOS arm64
Would be great to have all the tests like OrcaSlicer into Prusa itself.

DataPocalypse42, 2024-11-21:
Would be an awesome feature, sad that Prusa didn't even replied to this....

m10d, 2024-11-28:
respectfully @bubnikv and @josefprusa if you read this - the response given to @ibash 's WIP PR is disheartening and I humbly feel quite inappropriate.  I'm posting here directly, rather than over on [ibash 's original WIP thread,](https://github.com/prusa3d/PrusaSlicer/pull/7658) since OP there is indicated they are disengaged (it would seem, at least in part due to unresponsiveness of Prusa)

[from bubniv](https://github.com/prusa3d/PrusaSlicer/pull/7658#issuecomment-1004075400):

> Islam, thanks for your effort. However please understand that we would like to review the SuperSlicer calibration workflow internally before porting anything. It may happen that we will not be quite happy with the workflow and to provide it to our customers we would want to adjust it.
> 
> Second, somebody from our team will need to review all your implementation and we are busy with our own assignments. We are also now busy with planning of 2.5.0 features, thus we may be slow working with you.

bubniv makes two unrelated arguments here:

1. Prusa needs to understand the algo, and consider options from a design perspective, before allowing the feature into the codebase:  
    - This makes total sense, they would be the ones maintaining it, after all.  
    - But this isn't some obscure feature, as it's oft-requested and available not just in superslicer but many of the other major slicers as well.  
    - Thus this response is severely lacking in what I feel is reasonable detail.  Is this feature on Prusa's roadmap?  Surely for a feature this common _some_ thought has been put into it?  
3. Prusa employees are busy, it's too much time to review:  
    - IFF you want open source, and someone comes with actual code & real sustained effort, you owe them a meaningful response
    - of course I am not (and cannot) dictate to you if you actually want a FOSS community ... but if you do clearly some minimal engagement with folks _actually trying to contribute_ is going to be necessary.

I'd ask that a meaningful response be given, thank you.  (former mk3, current mk4/s, xl5t owner, and another frequent _needer of basic cal objects & temp towers_)

d-r-g, 2025-02-07:
Hello!  Just jumping on the comment list asking for this same feature.  I switched to Orca for a while and loved it.  I put prusalicer back on my computer when I reinstalled, and I miss that feature.  I'll probably move onto another slicer, but I'd prefer to stay with the one officially supported by Prusa.
Please consider adding calibration functionality.

ARCowboy, 2025-07-06:
Just piling on here... PrusaSlicer needs the same calibration tools as Orca:

1. Temperature Tower
2. Flow Rate
3. Pressure Advance
4. Retraction
5. VFA

I see some funny/snarky comments from Josef Prusa on reddit that we're free to use Orca, but can't complain when it doesn't work... I wouldn't *want* to use Orca if it didn't have more features, mainly an easy way to run calibrations and adjust pressure advance without inserting g-code.  

It is driving me away from Prusa, since I need to calibrate every filament yet have to do it manually.  I've had better results throwing off brand filaments in my Bambu printer with zero tuning than using the generic profiles in PrusaSlicer on a factory built MK4S...

DataPocalypse42, 2025-07-07:
> It is driving me away from Prusa, since I need to calibrate every filament yet have to do it manually. I've had better results throwing off brand filaments in my Bambu printer with zero tuning than using the generic profiles in PrusaSlicer on a factory built MK4S...

Which kind of filament brand/material require to calibrate every spools?

Aviator4x4, 2025-07-07:
> Which kind of filament brand/material require to calibrate every spools?

Not every spool, but different brands of filament.  Just off the top of my head I have prusament, bambu, overture, elegoo, sunlu, and colorfabb... I'm sure I'm forgetting a couple.  Some of them are "High Flow" PETG or TPU, some aren't.  Some are foaming TPU or PLA, which require in depth calibration per use case.  Usually differences in color haven't been a big deal, just brand/material profiles.

And that's where Prusa Slicer has been frustrating compared to Orca.

My usual workflow for a new filament, which is not uncommon:

0.  Dry filament
1.  Pick a stock profile (either from that brand or generic if none available).  
2.  Print Benchy.
3.  Assess if there are aspects that could be improved upon with calibration (Temp tower and flow ratio usually, sometimes PA).
4.  Reprint benchy or another "all in one, test everything" deal.
5.  Possibly repeat 2-4, but generally I am pretty happy with it at this step so I will save a "tuned" profile.

In Orca, it's 3-4 clicks and the calibration is printing.  In PrusaSlicer, it's much more involved.  Considering the recommendation pretty much every forum/reddit post gives, after drying filament and cleaning build plate, is to run calibrations... I am surprised they are not incorporated.

DataPocalypse42, 2025-07-07:
> My usual workflow for a new filament, which is not uncommon:
>
>     0. Dry filament
>
>     1. Pick a stock profile (either from that brand or generic if none available).
>
>     2. Print Benchy.
>
>     3. [...]

I've seen a lot of posts suggesting that kind of calibration set (temp tower and benchy), the prints are so small this will not reflect real print usage due to layer time. It was relevant when printers were slow, but with high speed printing you're better to print a real part and adjust temperatures depending on color/overhangs and same for the flow, you don't need a calibration but adjust if you see something wrong.

Even for pressure advance, this can be spotted easily if it's too high or low, but default values for the MK4S are very good and didn't require calibration ( I have polymaker ASA, Sunlu PETG, Prusament, 3Djake TPU, no one required to be fine tuned ).

Maybe for engineering filament, the full calibration set can be an insurance to avoid wasting material, but the temperature tower will still be wrong if you didn't print at the same speed

DataPocalypse42, 2025-07-07:
> I disagree with your opinion that calibration/Benchy prints are irrelevant/antiquated. Sure, the collective knowledge of Prusa forums/Reddit could be outdated folklore, but even Prusa CS themselves have suggested calibration prints when encountering issues.

I'm just saying that Benchy print test and temperature tower are useless, not all calibrations like Pressure Advance ( Most useful IMO ) and a flow rate « in case of issues ».

You don't need to do a 4hours print, but just find a print that reach the speed limits of your profile ( STRUCTURAL or SPEED ) that is useful and you can compare, because the benchy and speed tower will be limited by the layer time, you will never get the same result with a real world part.

They just setup the benchy on USB keys from factory because they know it will print beautiful, it's not a real benchmark test

closed 2026-09-29 by prusaslicer-bot[bot]